moononournation/Arduino_GFX #141: ESP32 S3

xmz0916, 2022-03-24:
Hi. When will ESP32 S3 be supported

moononournation, 2022-03-24:
I am also waiting for arduino-esp32 support ESP32S3

xmz0916, 2022-03-27:
謝謝，我使用Arduino的S3分支已經成功點亮NT35510了

closed 2022-03-27 by xmz0916

modi12jin, 2022-04-02:
@moononournation arduino esp32 has released preview version supporting S3

moononournation, 2022-04-02:
Don't worry too much, ESP32 S3 also implemented SPI, so Arduino_GFX can support it with HWSPI.

modi12jin, 2022-04-02:
@moononournation I'm making an S3 board, don't know when i80 and 16 will be supported?

moononournation, 2022-04-04:
@modi12jin checked ESP32S3 DataSheet, it should should able to use all ESP32S2PAR* class but not yet tested. Can you help me test it?

xmz0916, 2022-04-14:
> @modi12jin检查了 ESP32S3 数据表，它应该能够使用所有 ESP32S2PAR* 类，但尚未经过测试。你能帮我测试一下吗？

我在使用ESP32S3，直接通過複製ESP32S2PAR* 类，然後修改成S3的，可以正常使用，想請問下S3會支持LCD硬16位並口嗎？

xmz0916, 2022-04-14:
`#include "Arduino_DataBus.h"

#if CONFIG_IDF_TARGET_ESP32S3

#ifndef _ARDUINO_ESP32S3PAR16_H_
#define _ARDUINO_ESP32S3PAR16_H_

class Arduino_ESP32S3PAR16 : public Arduino_DataBus
{
public:
  Arduino_ESP32S3PAR16(int8_t dc, int8_t cs, int8_t wr, int8_t rd); // Constructor

  void begin(int32_t speed = 0, int8_t dataMode = 0) override;
  void beginWrite() override;
  void endWrite() override;
  void writeCommand(uint8_t) override;
  void writeCommand16(uint16_t) override;
  void write(uint8_t) override;
  void write16(uint16_t) override;
  void writeRepeat(uint16_t p, uint32_t len) override;
  void writePixels(uint16_t *data, uint32_t len) override;

  void writeC8D8(uint8_t c, uint8_t d) override;
  void writeC8D16(uint8_t c, uint16_t d) override;
  void writeC8D16D16(uint8_t c, uint16_t d1, uint16_t d2) override;
  void writeBytes(uint8_t *data, uint32_t len) override;
  void writePattern(uint8_t *data, uint8_t len, uint32_t repeat) override;

  void writeIndexedPixels(uint8_t *data, uint16_t *idx, uint32_t len) override;
  void writeIndexedPixelsDouble(uint8_t *data, uint16_t *idx, uint32_t len) override;

protected:
private:
  INLINE void WRITE(uint8_t d);
  INLINE void WRITE16(uint16_t d);
  INLINE void DC_HIGH(void);
  INLINE void DC_LOW(void);
  INLINE void CS_HIGH(void);
  INLINE void CS_LOW(void);

  int8_t _dc, _cs, _wr, _rd;

  PORTreg_t _dcPortSet; ///< PORT register SET
  PORTreg_t _dcPortClr; ///< PORT register CLEAR
  uint32_t _dcPinMask;  ///< Bitmask

  PORTreg_t _csPortSet; ///< PORT register SET
  PORTreg_t _csPortClr; ///< PORT register CLEAR
  uint32_t _csPinMask;  ///< Bitmask

  PORTreg_t _wrPortSet; ///< PORT register SET
  PORTreg_t _wrPortClr; ///< PORT register CLEAR
  uint32_t _wrPinMask;  ///< Bitmask

  PORTreg_t _rdPortSet; ///< PORT register SET
  PORTreg_t _rdPortClr; ///< PORT register CLEAR
  uint32_t _rdPinMask;  ///< Bitmask

  PORTreg_t _dataPortSet; ///< PORT register SET
  PORTreg_t _dataPortClr; ///< PORT register CLEAR
  uint32_t _dataClrMask;
};

#endif // _ARDUINO_ESP32S3PAR16_H_

#endif // #if CONFIG_IDF_TARGET_ESP32S3
`

xmz0916, 2022-04-14:
`#include "Arduino_ESP32S3PAR16.h"

#if CONFIG_IDF_TARGET_ESP32S3

Arduino_ESP32S3PAR16::Arduino_ESP32S3PAR16(int8_t dc, int8_t cs, int8_t wr, int8_t rd)
    : _dc(dc), _cs(cs), _wr(wr), _rd(rd)
{
}

void Arduino_ESP32S3PAR16::begin(int32_t speed, int8_t dataMode)
{
  pinMode(_dc, OUTPUT);
  digitalWrite(_dc, HIGH); // Data mode
  if (_dc >= 32)
  {
    _dcPinMask = digitalPinToBitMask(_dc);
    _dcPortSet = (PORTreg_t)&GPIO.out1_w1ts.val;
    _dcPortClr = (PORTreg_t)&GPIO.out1_w1tc.val;
  }
  else if (_dc != GFX_NOT_DEFINED)
  {
    _dcPinMask = digitalPinToBitMask(_dc);
    _dcPortSet = (PORTreg_t)&GPIO.out_w1ts;
    _dcPortClr = (PORTreg_t)&GPIO.out_w1tc;
  }

  if (_cs != GFX_NOT_DEFINED)
  {
    pinMode(_cs, OUTPUT);
    digitalWrite(_cs, HIGH); // disable chip select
  }
  if (_cs >= 32)
  {
    _csPinMask = digitalPinToBitMask(_cs);
    _csPortSet = (PORTreg_t)&GPIO.out1_w1ts.val;
    _csPortClr = (PORTreg_t)&GPIO.out1_w1tc.val;
  }
  else if (_cs != GFX_NOT_DEFINED)
  {
    _csPinMask = digitalPinToBitMask(_cs);
    _csPortSet = (PORTreg_t)&GPIO.out_w1ts;
    _csPortClr = (PORTreg_t)&GPIO.out_w1tc;
  }
  else
  {
    _csPinMask = 0;
    _csPortSet = _dcPortSet;
    _csPortClr = _dcPortClr;
  }

  pinMode(_wr, OUTPUT);
  digitalWrite(_wr, HIGH); // Set write strobe high (inactive)
  if (_wr >= 32)
  {
    _wrPinMask = digitalPinToBitMask(_wr);
    _wrPortSet = (PORTreg_t)&GPIO.out1_w1ts.val;
    _wrPortClr = (PORTreg_t)&GPIO.out1_w1tc.val;
  }
  else if (_wr != GFX_NOT_DEFINED)
  {
    _wrPinMask = digitalPinToBitMask(_wr);
    _wrPortSet = (PORTreg_t)&GPIO.out_w1ts;
    _wrPortClr = (PORTreg_t)&GPIO.out_w1tc;
  }

  if (_rd != GFX_NOT_DEFINED)
  {
    pinMode(_rd, OUTPUT);
    digitalWrite(_rd, HIGH);
  }
  if (_rd >= 32)
  {
    _rdPinMask = digitalPinToBitMask(_rd);
    _rdPortSet = (PORTreg_t)&GPIO.out1_w1ts.val;
    _rdPortClr = (PORTreg_t)&GPIO.out1_w1tc.val;
  }
  else if (_rd != GFX_NOT_DEFINED)
  {
    _rdPinMask = digitalPinToBitMask(_rd);
    _rdPortSet = (PORTreg_t)&GPIO.out_w1ts;
    _rdPortClr = (PORTreg_t)&GPIO.out_w1tc;
  }
  else
  {
    _rdPinMask = 0;
    _rdPortSet = _dcPortSet;
    _rdPortClr = _dcPortClr;
  }

  pinMode(0, OUTPUT);
  pinMode(1, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, OUTPUT);
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);
  pinMode(14, OUTPUT);
  pinMode(15, OUTPUT);

  // INIT 16-bit mask
  _dataClrMask = 0xFFFF;
  _dataPortSet = (PORTreg_t)&GPIO.out_w1ts;
  _dataPortClr = (PORTreg_t)&GPIO.out_w1tc;
  *_dataPortClr = _dataClrMask;
}

void Arduino_ESP32S3PAR16::beginWrite()
{
  DC_HIGH();
  CS_LOW();
}

void Arduino_ESP32S3PAR16::endWrite()
{
  CS_HIGH();
}

void Arduino_ESP32S3PAR16::writeCommand(uint8_t c)
{
  DC_LOW();

  WRITE(c);

  DC_HIGH();
}

void Arduino_ESP32S3PAR16::writeCommand16(uint16_t c)
{
  DC_LOW();

  WRITE16(c);

  DC_HIGH();
}

void Arduino_ESP32S3PAR16::write(uint8_t d)
{
  WRITE(d);
}

void Arduino_ESP32S3PAR16::write16(uint16_t d)
{
  WRITE16(d);
}

void Arduino_ESP32S3PAR16::writeRepeat(uint16_t p, uint32_t len)
{
  *_dataPortClr = _dataClrMask;
  *_dataPortSet = p;
  while (len--)
  {
    *_wrPortClr = _wrPinMask;
    *_wrPortSet = _wrPinMask;
  }
}

void Arduino_ESP32S3PAR16::writePixels(uint16_t *data, uint32_t len)
{
  while (len--)
  {
    *_dataPortClr = _dataClrMask;
    *_dataPortSet = *data++;
    *_wrPortClr = _wrPinMask;
    *_wrPortSet = _wrPinMask;
  }
}

void Arduino_ESP32S3PAR16::writeC8D8(uint8_t c, uint8_t d)
{
  DC_LOW();

  WRITE(c);

  DC_HIGH();

  WRITE(d);
}

void Arduino_ESP32S3PAR16::writeC8D16(uint8_t c, uint16_t d)
{
  DC_LOW();

  WRITE(c);

  DC_HIGH();

  _data16.value = d;
  WRITE(_data16.msb);
  WRITE(_data16.lsb);
}

void Arduino_ESP32S3PAR16::writeC8D16D16(uint8_t c, uint16_t d1, uint16_t d2)
{
  DC_LOW();

  WRITE(c);

  DC_HIGH();

  _data16.value = d1;
  WRITE(_data16.msb);
  WRITE(_data16.lsb);

  _data16.value = d2;
  WRITE(_data16.msb);
  WRITE(_data16.lsb);
}

void Arduino_ESP32S3PAR16::writeBytes(uint8_t *data, uint32_t len)
{
  while (len > 1)
  {
    _data16.msb = *data++;
    _data16.lsb = *data++;
    WRITE16(_data16.value);
    len -= 2;
  }
  if (len)
  {
    WRITE(*data);
  }
}

void Arduino_ESP32S3PAR16::writePattern(uint8_t *data, uint8_t len, uint32_t repeat)
{
  while (repeat--)
  {
    writeBytes(data, len);
  }
}

void Arduino_ESP32S3PAR16::writeIndexedPixels(uint8_t *data, uint16_t *idx, uint32_t len)
{
  while (len--)
  {
    *_dataPortClr = _dataClrMask;
    *_dataPortSet = idx[*data++];
    *_wrPortClr = _wrPinMask;
    *_wrPortSet = _wrPinMask;
  }
}

void Arduino_ESP32S3PAR16::writeIndexedPixelsDouble(uint8_t *data, uint16_t *idx, uint32_t len)
{
  while (len--)
  {
    *_dataPortClr = _dataClrMask;
    *_dataPortSet = idx[*data++];
    *_wrPortClr = _wrPinMask;
    *_wrPortSet = _wrPinMask;
    *_wrPortClr = _wrPinMask;
    *_wrPortSet = _wrPinMask;
  }
}

INLINE void Arduino_ESP32S3PAR16::WRITE(uint8_t d)
{
  *_dataPortClr = _dataClrMask;
  *_dataPortSet = d;
  *_wrPortClr = _wrPinMask;
  *_wrPortSet = _wrPinMask;
}

INLINE void Arduino_ESP32S3PAR16::WRITE16(uint16_t d)
{
  *_dataPortClr = _dataClrMask;
  *_dataPortSet = d;
  *_wrPortClr = _wrPinMask;
  *_wrPortSet = _wrPinMask;
}

/******** low level bit twiddling **********/

INLINE void Arduino_ESP32S3PAR16::DC_HIGH(void)
{
  *_dcPortSet = _dcPinMask;
}

INLINE void Arduino_ESP32S3PAR16::DC_LOW(void)
{
  *_dcPortClr = _dcPinMask;
}

INLINE void Arduino_ESP32S3PAR16::CS_HIGH(void)
{
  *_csPortSet = _csPinMask;
}

INLINE void Arduino_ESP32S3PAR16::CS_LOW(void)
{
  *_csPortClr = _csPinMask;
}

#endif // #if CONFIG_IDF_TARGET_ESP32S3
`

moononournation, 2022-04-14:
no need duplicate class, latest version can direct use ESP32S2PAR* class in ESP32S3

modi12jin, 2022-04-23:
[image]

```
  pinMode(37, OUTPUT);
  pinMode(36, OUTPUT);
  pinMode(35, OUTPUT);
  pinMode(34, OUTPUT);
  pinMode(33, OUTPUT);
  pinMode(48, OUTPUT);
  pinMode(21, OUTPUT);
  pinMode(18, OUTPUT);
  pinMode(17, OUTPUT);
  pinMode(39, OUTPUT);
  pinMode(38, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(12, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(10, OUTPUT);
```

```
#include <Arduino_GFX_Library.h>

#define TFT_CS -1
#define TFT_DC 16
#define TFT_RST -1
#define TFT_RD -1
#define TFT_WR 15 
#define TFT_BL 5

Arduino_DataBus *bus = new Arduino_ESP32S2PAR16(TFT_DC,TFT_CS,TFT_WR,TFT_RD);

Arduino_GFX* gfx = new Arduino_R61529(bus, TFT_RST, 0 /* rotation */, true /* IPS */);
```

[image]

@moononournation I tested the screen on ESP32S3 and it didn't display properly

moononournation, 2022-04-23:
ESP32S2PAR16 using gpio 0-15 for parallel port, and also please ensure the LCD RD pin has pulled-up.

modi12jin, 2022-04-24:
| R61529引脚编号 |     LCD功能     | ESP32-S3引脚 |
| :------------: | :-------------: | :----------: |
|       1        |       XL        |     悬空     |
|       2        |       YU        |     悬空     |
|       3        |       XR        |     悬空     |
|       4        |       YD        |     悬空     |
|       5        |       GND       |     GND      |
|       6        | IOVCC(1.8/2.8V) |     3.3V     |
|       7        |    VCI(2.8V)    |     3.3V     |
|       8        |      FMARK      |     悬空     |
|       9        |    CS/SPI CS    |     GND      |
|       10       |      RS/DC      |     IO16     |
|       11       |       WR        |     IO15     |
|       12       |       RD        |     3.3V     |
|       13       |   SPI SDI/SDA   |     悬空     |
|       14       |     SPI SDO     |     悬空     |
|       15       |      RESET      |     3.3V     |
|       16       |       GND       |     GND      |
|       17       |       DB0       |     IO37     |
|       18       |       DB1       |     IO36     |
|       19       |       DB2       |     IO35     |
|       20       |       DB3       |     IO34     |
|       21       |       DB4       |     IO33     |
|       22       |       DB5       |     IO48     |
|       23       |       DB6       |     IO21     |
|       24       |       DB7       |     IO18     |
|       25       |       DB8       |     IO17     |
|       26       |       DB9       |     IO39     |
|       27       |      DB10       |     IO38     |
|       28       |      DB11       |     IO2      |
|       29       |      DB12       |     IO7      |
|       30       |      DB13       |     IO12     |
|       31       |      DB14       |     IO6      |
|       32       |      DB15       |     IO10     |
|       33       |        A        |     3.3V     |
|     34~36      |        K        | IO5(LED_PWM) |
|       37       |       GND       |     GND      |
|       38       |       IM0       |     3.3V     |
|       39       |       IM1       |     GND      |
|       40       |       IM2       |     3.3V     |

[image]

@moononournation I can't change the physical pins as it's a homemade PCB. So what to do, change the default pins of the library?

There is no problem with the hardware, which can be confirmed by this picture.

[image]

moononournation, 2022-04-24:
try ESP32PAR16QQ

modi12jin, 2022-04-26:
```
#include <Arduino_GFX_Library.h>

#define TFT_CS -1
#define TFT_DC 16
#define TFT_RST -1
#define TFT_RD -1
#define TFT_WR 15 
#define TFT_BL 5
#define D0 37
#define D1 36
#define D2 35
#define D3 34
#define D4 33
#define D5 48
#define D6 21
#define D7 18
#define D8 17
#define D9 39
#define D10 38
#define D11 2
#define D12 7
#define D13 12
#define D14 6
#define D15 10

Arduino_DataBus *bus = new Arduino_ESP32PAR16(TFT_DC,TFT_CS,TFT_WR,TFT_RD,D0,D1,D2,D3,D4,D5,D6,D7,D8,D9,D10,D11,D12,D13,D14,D15);

Arduino_GFX* gfx = new Arduino_R61529(bus, TFT_RST, 0 /* rotation */, true /* IPS */);
```

@moononournation Can not work normally

moononournation, 2022-04-27:
Committed a quick fix on Arduino_ESP32PAR16, please get update to test.
In long term, should implement an ESP32i80PAR16 data bus for ESP32S3.

moononournation, 2022-05-14:
Now you can use Arduino_ESP32LCD16 in ESP32S3.

xmz0916, 2022-05-18:
Look forward to the latest version of 1.2.3 or 1.3