moononournation/Arduino_GFX #142: 在使用ESP32控制NT35510測試LVGL出現問題

xmz0916, 2022-03-27:
開啟了FPS、CPU和內存使用量提示時，其顯示的內容不全，並且在使用組件時，組件顯示不全，會出現gfx->fillScreen(RED)設置的背景顏色，測試了其他幾款屏幕都沒有問題，僅在NT35510上出現過。使用Arduino_GFX的繪圖函數都很正常，唯獨使用lvgl出現顯示問題

xmz0916, 2022-03-27:
顯示錯誤的問題，我把圖片放到了這裡
https://www.xutoubee.cn/wp-content/uploads/2022/03/1.jpg
https://www.xutoubee.cn/wp-content/uploads/2022/03/2.jpg

xmz0916, 2022-03-27:
`#include <lvgl.h>
#include <Arduino_GFX_Library.h>
#include "ui_helpers.h"
#include "ui.h"

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_ESP32S3PAR16(16 /* DC */, 17 /* CS */, 18 /* WR */, 20 /* RD */);

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_NT35510(bus, 21 /* RST */, 1 /* rotation */);

/* Change to your screen resolution */
static uint32_t screenWidth;
static uint32_t screenHeight;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t *disp_draw_buf_1;
static lv_color_t *disp_draw_buf_2;
static lv_disp_drv_t disp_drv;

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
   uint32_t w = (area->x2 - area->x1 + 1);
   uint32_t h = (area->y2 - area->y1 + 1);

   gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);

   lv_disp_flush_ready(disp);
}

void setup()
{
   Serial.begin(115200);
   // while (!Serial);
   Serial.println("LVGL Hello World");

   // Init Display
   gfx->begin();
   gfx->fillScreen(RED);

   lv_init();

   screenWidth  = 800;
   screenHeight = 480;
   disp_draw_buf_1 = (lv_color_t *)malloc(sizeof(lv_color_t) * screenWidth * 10);
   disp_draw_buf_2 = (lv_color_t *)malloc(sizeof(lv_color_t) * screenHeight * 10);
   
   lv_disp_draw_buf_init(&draw_buf, disp_draw_buf_1, disp_draw_buf_2, screenWidth * 10);

      /* Initialize the display */
      lv_disp_drv_init(&disp_drv);
      /* Change the following line to your display resolution */
      disp_drv.hor_res = screenWidth;
      disp_drv.ver_res = screenHeight;
      disp_drv.flush_cb = my_disp_flush;
      disp_drv.draw_buf = &draw_buf;
      lv_disp_drv_register(&disp_drv);

     

      /* Create simple label */
      ui_init(); 
}

void loop()
{
   lv_timer_handler(); /* let the GUI do its work */
   lv_label_set_text_fmt(ui_Label2, "Num:%d",random(100));
   delay(300);
}`

moononournation, 2022-03-27:
No idea, is it related to screen size? try narrow down the size to check:
```
screenWidth = 320;
screenHeight = 240;
```

xmz0916, 2022-03-27:
試過了，都沒有用

moononournation, 2022-04-23:
I also found similar effect while testing ILI9806. The latest LVGL example code enforced ESP32 use internal memory for the buffer, please check is it can fix your issue.

xmz0916, 2022-04-27:
> 

Thank you. The previous problem has been solved