moononournation/Arduino_GFX #144: Cant make this work with STM32F411xx

dungar43, 2022-03-29:
I cant make this work with STM32F411xx.  The supported devices seems to include the WeAct BlackPill V2.0 (BlackPill F411CE) which I used.

moononournation, 2022-03-29:
can not compile? or no display? can you show your connection and code?

dungar43, 2022-03-30:
Seems to compile ok but no activity on any pins. I did not connect the
display because I could not tell how to determine what STM32 pins to use.
I’m new to this. Thanks

moononournation, 2022-03-30:
https://www.instructables.com/ArduinoGFX/

moononournation, 2022-03-31:
you should change the display class to your own display:
https://github.com/moononournation/Arduino_GFX/wiki/Display-Class

closed 2022-04-05 by moononournation

dungar43, 2022-04-28:
Thanks for the help. Just got back on this. I purchased an IL9341 and it
works fine
/* More dev device declaration:
https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration
*/
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class:
https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = create_default_Arduino_DataBus();

/* More display class:
https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 1/* rotation */,
false /* IPS */);

#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
I need to modify the connections from the STM32F411 to the IL9341 to match
another hardware setup. I assume its something like

Arduino_DataBus *bus = create_default_Arduino_DataBus();

but I dont see how to do it. Any help will be appreciated.