moononournation/Arduino_GFX #151: Help with Waveshare Pico Eval Kit configuration

hexcode0x, 2022-04-23:
I think I'm missing something basic.  I have the Waveshare Pico Eval Kit described here: https://www.waveshare.com/pico-eval-board.htm

I can get the LCD to work using demo code, but I wanted to try your Arduino_GFX lib to see if I can get the 3-bit / 8 color mode working.  But first, I'm trying to get the 16-bit color mode to work.

I started with the PDQgraphicstest example and modified these lines, but no LCD display.  I am getting all the serial output indicating it is running the graphics test.  Any ideas on what I could have done wrong?  I can confirm the pin numbers on the Waveshare page are correct as they work in the demo code.

#elif defined(ARDUINO_RASPBERRY_PI_PICO)
#define TFT_CS 9 // GFX_NOT_DEFINED for display without CS pin
#define TFT_DC 8
#define TFT_RST 15
#define TFT_BL 13

Arduino_DataBus *bus = new Arduino_RPiPicoSPI(TFT_DC, TFT_CS, 10 /* SCK */, 11 /* MOSI */, 12 /* MISO */, spi1 /* spi */);

// ILI9488 8-bit LCD 320x480
Arduino_GFX *gfx = new Arduino_ILI9488(bus, TFT_RST, 0 /* rotation */, false /* IPS */);

Thanks!

moononournation, 2022-04-23:
for SPI, you should use 18bit class:
```C
Arduino_GFX *gfx = new Arduino_ILI9488_18bit(bus, TFT_RST, 0 /* rotation */, false /* IPS */);
```

hexcode0x, 2022-04-23:
Just tried that, no change.  Does it look like I'm supplying the right pins according to the board description?  I double checked but maybe I'm not understand the way the Pico Arduino layer works.

Also, curiously, the Waveshare demo code does use 16-bit SPI transfers for the ILI9488 and it does work.  The latest datasheet I can find does say that is supported.

moononournation, 2022-04-23:
no, according to waveshare provided DataSheet, SPI only can be 3-bit and 18-bit:
https://www.waveshare.net/w/upload/2/2d/ILI9488_Data_Sheet.pdf

moononournation, 2022-04-23:
checked their init code, wild guess it is actually ST7796. You may change the display class to check this out.

hexcode0x, 2022-04-23:
Ok, I think I figured out what is going on.  They don't advertise this, but it looks like they have the ILI9488 configured in 16 bit parallel mode with a complicated spi to parallel bus converter using 2 74hc4094 chips, a 74hc4040 and a 74hc04d.  This is a very bad design.

closed 2022-04-27 by hexcode0x