moononournation/Arduino_GFX #156: not declared in scope error

CodeOn-ArK, 2022-05-15:
Hi,
When I am compiling SPIFFS_GIF example, It's giving the following error

```

C:\Users\KIIT\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp: In function 'Arduino_DataBus* create_default_Arduino_DataBus()':
C:\Users\KIIT\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:10:55: error: 'DF_GFX_SCK' was not declared in this scope
     return new Arduino_ESP32SPI(DF_GFX_DC, DF_GFX_CS, DF_GFX_SCK, DF_GFX_MOSI, DF_GFX_MISO);
                                                       ^
C:\Users\KIIT\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:10:67: error: 'DF_GFX_MOSI' was not declared in this scope
     return new Arduino_ESP32SPI(DF_GFX_DC, DF_GFX_CS, DF_GFX_SCK, DF_GFX_MOSI, DF_GFX_MISO);
                                                                   ^
C:\Users\KIIT\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:10:80: error: 'DF_GFX_MISO' was not declared in this scope
     return new Arduino_ESP32SPI(DF_GFX_DC, DF_GFX_CS, DF_GFX_SCK, DF_GFX_MOSI, DF_GFX_MISO);
                                                                                ^

```

moononournation, 2022-05-15:
what dev board are you using and selected in Arduino IDE?

CodeOn-ArK, 2022-05-18:
esp32 dev board with 4MB flash, When I include the macros in core header files, it works

CodeOn-ArK, 2022-05-18:
It works great!! thanks a lot!

closed 2022-05-18 by moononournation

moononournation, 2022-06-07:
- ensure upgraded latest Arduino IDE
- ensure Arduino/libraries folder only have either of "Arduino_GFX" or "GFX Library for Arduino"