moononournation/Arduino_GFX #157: No image diplayed on screen

XavierColombel, 2022-05-17:
I tried your [WiFiPhotoFrame](https://github.com/moononournation/Arduino_GFX/tree/master/examples/WiFiPhotoFrame) example but I use my own server to display a simple image.

Here it is:

```
const express = require("express");
const app = new express();

app.get("/", (req, res) => {
  res.sendFile("/Users/xavier/Github/Tests/testServer/image.jpg");
});

app.listen(3333, () => {
  console.log("server is up on PORT 3333 !");
});
```

Actually, my screen displays informations about the download, but still no image. Here is a screen:
[image]

Could you tell me if the `jpegDec.decode(JPG_SCALE_NONE, jpegDec.gfx_writer, gfx);` function is working as it should or if I missed something? Did I send something wrong (`res.sendFile`)? Should I send the image in another format?

Thanks.

moononournation, 2022-05-17:
just ensure the JPEG file is not progressive encoded. you may test more file to check it.

moononournation, 2022-05-17:
and seems you web server do not read the parameter to resize the file, please ensure the image is smaller than 128x128 to fit your screen.

closed 2022-05-18 by XavierColombel