moononournation/Arduino_GFX #161: Strange issue

m-menon, 2022-05-23:
hi,

Just tried the library with an ESP32 + R61529. I thoroughly checked the wiring a few times to ensure that the connections to the TFT are correct. I ran a simple HelloWorld sketch without any modification - except initializing the GFX to ESP32 parallel 8 bit. My test code

```
#include <Arduino_GFX_Library.h>
#define GFX_BL 0 // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

#define PROTO_TFT_CS 32
#define PROTO_TFT_DC 5
#define PROTO_TFT_RST 33
#define PROTO_TFT_WR 2
#define PROTO_TFT_RD 4

#define PROTO_TFT_D0 18
#define PROTO_TFT_D1 19
#define PROTO_TFT_D2 21
#define PROTO_TFT_D3 22
#define PROTO_TFT_D4 23
#define PROTO_TFT_D5 25
#define PROTO_TFT_D6 26
#define PROTO_TFT_D7 27

// ESP32 parallel 8-bit
Arduino_DataBus *bus = new Arduino_ESP32PAR8(PROTO_TFT_DC, PROTO_TFT_CS, PROTO_TFT_WR /* WR */, PROTO_TFT_RD /* RD */, PROTO_TFT_D0 /* D0 */, PROTO_TFT_D1 /* D1 */, PROTO_TFT_D2 /* D2 */, PROTO_TFT_D3 /* D3 */, PROTO_TFT_D4 /* D4 */, PROTO_TFT_D5 /* D5 */, PROTO_TFT_D6 /* D6 */, PROTO_TFT_D7 /* D7 */);
Arduino_GFX *gfx = new Arduino_R61529(bus, PROTO_TFT_RST /* RST */, 2 /* rotation */, true /* IPS */);

void setup(void)
{
  gfx->begin();
  gfx->fillScreen(BLACK);

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
#endif

  gfx->setCursor(10, 10);
  gfx->setTextColor(RED);
  gfx->println("Hello World!");

  delay(5000); // 5 seconds
}

void loop()
{
  gfx->setCursor(random(gfx->width()), random(gfx->height()));
  gfx->setTextColor(random(0xffff), random(0xffff));
  gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
  gfx->println("Hello World!");

  delay(1000); // 1 second
}
```

Here's a [video](https://drive.google.com/file/d/1KSq4BdbX2BY7BHtSsKSg1yTZAzhCXrnF/view?usp=sharing) of what happens next:
https://drive.google.com/file/d/1KSq4BdbX2BY7BHtSsKSg1yTZAzhCXrnF/view?usp=sharing

Here's a picture of my board (R61529). I soldered the IM0, IM1 and IM2 pads to PARALLEL 8 bit. Not sure what could be wrong. Appreciate any pointers or help on this issue.

[image]
[image]

I did not use any of the reserved esp32 pins. Also I did not enable the PSRAM or use the PSRAM. I even tried different orientation / rotation. Thanks.

moononournation, 2022-06-09:
https://user-images.githubusercontent.com/13228891/172877195-3c19ebca-0fab-48bc-a702-8ed5f1cbc62b.MOV
just checked latest version can run well with R61529 ESP32PAR8

moononournation, 2022-06-09:
my pin setting is:
```
Arduino_DataBus *bus = new Arduino_ESP32PAR8(
     32 /* DC */, GFX_NOT_DEFINED /* CS */, 21 /* WR */, GFX_NOT_DEFINED /* RD */,
     19 /* D0 */, 23 /* D1 */, 18 /* D2 */, 5 /* D3 */, 17 /* D4 */, 16 /* D5 */, 25 /* D6 */, 26 /* D7 */
);
Arduino_GFX *gfx = new Arduino_R61529(bus, 33 /* RST */, 0 /* rotation */, true /* IPS */);
```

mm108, 2022-06-09:
Thanks for the pointers. I'll check that out.

closed 2022-06-09 by mm108