moononournation/Arduino_GFX #163: Issue with displaying GIFs using the ImageViewerAnimatedGIF example

TripleT-circuitmaker, 2022-05-27:
Hi!

I’m trying to display a gif on a device that has 64 kb SRAM on a 240 x 240 screen. I can’t play gifs larger than 64 x 64. And the gifs that are 64 x 64 don’t loop. I think this is because I don’t have sufficient memory, is that correct? 

Is there any way around this?

Any help would be appreciated!

moononournation, 2022-05-27:
yes, the example decoder requires full frame memory. you may try to find another GIF decoder that can decode block by block.

TripleT-circuitmaker, 2022-05-27:
Do you know of any of GIF decoder libraries that you can recommend?

TripleT-circuitmaker, 2022-05-27:
Would this work for my application? 

[AnimatedGIF](https://github.com/bitbank2/AnimatedGIF)

moononournation, 2022-05-27:
Em... my actual recommendation is: if you want to play large Animated GIF, please use a dev board with more memory.

moononournation, 2022-05-27:
> Would this work for my application? 
> 
> 
> 
> [AnimatedGIF](https://github.com/bitbank2/AnimatedGIF)

yes you can try this

closed 2022-06-08 by moononournation