moononournation/Arduino_GFX #166: drawArcHelper & drawEllipseHelper functions not working

saturnhades, 2022-06-02:
I tried to use drawArcHelper() function, but nothing happened, only the text displayed not the ellipse helper work. Could you tell the problem or guide me on how to use that function? Below is the code I have used.
`#include <Arduino_GFX_Library.h>
Arduino_DataBus *bus = new Arduino_ESP32SPI(27 /* DC */, 5 /* CS */, 18 /* SCK */, 23 /* MOSI */, -1 /* MISO */, VSPI /* spi_num */);
Arduino_GFX *lcd = new Arduino_GC9A01(bus, 33 /* RST */, 0 /* rotation */, true /* IPS */);

void setup(){
  pinMode(22, OUTPUT);
  digitalWrite(22, HIGH);
  lcd->begin();
  delay(1000);
  lcd->fillScreen(BLACK);
  
}
void loop(){
  lcd->setCursor(100, 100);
  lcd->setTextColor(RED);
  lcd->println("Hello World!");
  delay(1000);
  lcd->fillScreen(BLACK);
  //lcd->fillArc(120,120,100,90,90,270,CYAN);
  lcd->fillArcHelper(120,120,90,30,90,2700,CYAN);
  lcd->drawEllipseHelper(120,120,20,30,2,YELLOW);
  delay(1000);
}
[image]
`

moononournation, 2022-06-02:
The helper function is for internal use. It only work within the calling of startWrite() and endWrite().

closed 2022-06-02 by moononournation

saturnhades, 2022-06-02:
Thanks got that, but I also have a request. Could you add the attribute to make the ends of the arc round-rectangle? It would give some nice look if you can then please help or if possible provide me relatable links or lines of code so that I can try.