moononournation/Arduino_GFX #172: ILI9486 wrong colors, wrong saturation or gamma overdrive

ojaksch, 2022-06-13:
I did some tests with the ILI9486 driver (parallel TFT) to load JPG pictures. This is working so far, but the pictures are looking horrible.
Have a look to some examples...left side is your Arduino_GFX, right side TFT_eSPI, both with the same ESP32 and display.
With an ILI9341 all is good.

[image]
[image]
[image]
[image]

moononournation, 2022-06-13:
ok, just align init code with TFT_eSPI. I do not have ILI9486 parallel display in hand, please help to check is it work.

moononournation, 2022-06-14:
mirrored? but you display jpeg not mirrored🤔

ojaksch, 2022-06-14:
> mirrored? but you display jpeg not mirroredthinking

No, jpeg is the original file like shown at top. But that doesn't matter, we had the same thing in #155 and I have fixed the correct rotation values for the ILI9486 by borrowing them from TFT_eSPI :)
All seems perfect now. Many thanks for restructuring this driver!

```
--- Arduino_ILI9486.cpp.org     2022-06-13 18:50:31.191415789 +0200
+++ Arduino_ILI9486.cpp 2022-06-14 07:33:54.834262354 +0200
@@ -26,16 +26,16 @@
   switch (_rotation)
   {
   case 1:
-    r = (ILI9486_MADCTL_MY | ILI9486_MADCTL_MV | ILI9486_MADCTL_BGR);
+    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MV);
     break;
   case 2:
-    r = (ILI9486_MADCTL_BGR);
+    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MX | ILI9486_MADCTL_MY);
     break;
   case 3:
     r = (ILI9486_MADCTL_MX | ILI9486_MADCTL_MV | ILI9486_MADCTL_BGR);
     break;
   default: // case 0:
-    r = (ILI9486_MADCTL_MX | ILI9486_MADCTL_MY | ILI9486_MADCTL_BGR);
+    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MX);
     break;
   }
   _bus->beginWrite();
```

[image]

amirjak0, 2022-11-13:
/**************************************************************************/
void Arduino_ILI9486::setRotation(uint8_t r)
{
  Arduino_TFT::setRotation(r);
   switch (_rotation)
   {
   case 1:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MV);
     break;
   case 2:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MX | ILI9486_MADCTL_MY);
     break;
   case 3:
     r = (ILI9486_MADCTL_MX | ILI9486_MADCTL_MV | ILI9486_MADCTL_BGR);
     break;
   default: // case 0:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MX);
     break;
   }
  _bus->beginWrite();
  _bus->writeC8D8(ILI9486_MADCTL, r);
  _bus->endWrite();
}

amirjak0, 2022-11-13:
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::writeAddrWindow(short, short, unsigned short, unsigned short)':
C:\Users\ali\Documents\Arduino\libraries\GFX_Library_for_Arduino-1.2.8\src\display/Arduino_ILI9486.h:63: multiple definition of `Arduino_ILI9486::writeAddrWindow(short, short, unsigned short, unsigned short)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:C:\Users\ali\Documents\Arduino\libraries\GFX_Library_for_Arduino-1.2.8\src\display/Arduino_ILI9486 - Copy - Copy.cpp:47: first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::begin(int)':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI94865beginEi+0x0): multiple definition of `Arduino_ILI9486::begin(int)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI94865beginEi+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `non-virtual thunk to Arduino_ILI9486::begin(int)':
Arduino_ILI9486.cpp:(.text._ZThn8_N15Arduino_ILI94865beginEi+0x0): multiple definition of `non-virtual thunk to Arduino_ILI9486::begin(int)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZThn8_N15Arduino_ILI94865beginEi+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::setRotation(unsigned char)':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI948611setRotationEh+0x0): multiple definition of `Arduino_ILI9486::setRotation(unsigned char)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI948611setRotationEh+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::invertDisplay(bool)':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI948613invertDisplayEb+0x0): multiple definition of `Arduino_ILI9486::invertDisplay(bool)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI948613invertDisplayEb+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::displayOn()':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI94869displayOnEv+0x0): multiple definition of `Arduino_ILI9486::displayOn()'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI94869displayOnEv+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::displayOff()':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI948610displayOffEv+0x0): multiple definition of `Arduino_ILI9486::displayOff()'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI948610displayOffEv+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::tftInit()':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI94867tftInitEv+0x0): multiple definition of `Arduino_ILI9486::tftInit()'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI94867tftInitEv+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::Arduino_ILI9486(Arduino_DataBus*, signed char, unsigned char, bool)':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI9486C2EP15Arduino_DataBusahb+0x0): multiple definition of `Arduino_ILI9486::Arduino_ILI9486(Arduino_DataBus*, signed char, unsigned char, bool)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI9486C2EP15Arduino_DataBusahb+0x0): first defined here
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486.cpp.o: In function `Arduino_ILI9486::Arduino_ILI9486(Arduino_DataBus*, signed char, unsigned char, bool)':
Arduino_ILI9486.cpp:(.text._ZN15Arduino_ILI9486C2EP15Arduino_DataBusahb+0x0): multiple definition of `Arduino_ILI9486::Arduino_ILI9486(Arduino_DataBus*, signed char, unsigned char, bool)'
libraries\GFX_Library_for_Arduino-1.2.8\display\Arduino_ILI9486 - Copy - Copy.cpp.o:Arduino_ILI9486 - Copy - Copy.cpp:(.text._ZN15Arduino_ILI9486C2EP15Arduino_DataBusahb+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
 Used: C:\Users\ali\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

moononournation, 2022-11-13:
I think you should remove you copy copy files first.

amirjak0, 2022-11-13:
i  remove  copy copy

Sketch uses 296806 bytes (22%) of program storage space. Maximum is 1310720 bytes.
Global variables use 32264 bytes (9%) of dynamic memory, leaving 295416 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.0-dev
Serial port COM3
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:4b:ec:0c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 840.2 kbit/s)...
Hash of data verified.
Compressed 17120 bytes to 11164...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11164 compressed) at 0x00001000 in 0.2 seconds (effective 674.7 kbit/s)...
Hash of data verified.
Compressed 296928 bytes to 153114...
Writing at 0x00010000... (10 %)
Writing at 0x00014000... (20 %)
Writing at 0x00018000... (30 %)
Writing at 0x0001c000... (40 %)
Writing at 0x00020000... (50 %)
Writing at 0x00024000... (60 %)
Writing at 0x00028000... (70 %)
Writing at 0x0002c000... (80 %)
Writing at 0x00030000... (90 %)
Writing at 0x00034000... (100 %)
Wrote 296928 bytes (153114 compressed) at 0x00010000 in 2.8 seconds (effective 863.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 396.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

**### but**

[image]

amirjak0, 2022-11-13:
Arduino_DataBus *bus = new Arduino_ESP32PAR8(15 /* DC */, 33 /* CS */, 4 /* WR */, 2 /* RD */,
    12 /* D0 */, 13 /* D1 */, 26 /* D2 */, 25 /* D3 */, 17 /* D4 */, 16 /* D5 */, 27 /* D6 */, 14 /* D7 */);

Arduino_GFX *gfx = new Arduino_ILI9486(bus, DF_GFX_RST, 3 /* rotation */, false /* IPS */);

ojaksch, 2022-11-14:
I see that you are using "rotation 3". Could you please re-test with "rotation 1"? I haven't tested nor touched this case...

ojaksch, 2022-11-14:
Yep, definition is wrong for case 3 and should be
```
  case 3:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MV | ILI9486_MADCTL_MX | ILI9486_MADCTL_MY);
    break;
```

ojaksch, 2022-11-14:
Re-Tested all four directions and this the (for me) working Arduino_ILI9486.cpp

```
/*
 * start rewrite from:
 * https://github.com/nopnop2002/esp-idf-parallel-tft
 */
#include "Arduino_ILI9486.h"

Arduino_ILI9486::Arduino_ILI9486(Arduino_DataBus *bus, int8_t rst, uint8_t r, bool ips)
    : Arduino_TFT(bus, rst, r, ips, ILI9486_TFTWIDTH, ILI9486_TFTHEIGHT, 0, 0, 0, 0)
{
}

void Arduino_ILI9486::begin(int32_t speed)
{
  Arduino_TFT::begin(speed);
}

/**************************************************************************/
/*!
    @brief   Set origin of (0,0) and orientation of TFT display
    @param   m  The index for rotation, from 0-3 inclusive
*/
/**************************************************************************/
void Arduino_ILI9486::setRotation(uint8_t r)
{
  Arduino_TFT::setRotation(r);
  switch (_rotation)
  {
  case 1:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MV);
    break;
  case 2:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MY);
    break;
  case 3:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MV | ILI9486_MADCTL_MX | ILI9486_MADCTL_MY);
    break;
  default: // case 0:
    r = (ILI9486_MADCTL_BGR | ILI9486_MADCTL_MX);
    break;
  }
  _bus->beginWrite();
  _bus->writeC8D8(ILI9486_MADCTL, r);
  _bus->endWrite();
}

void Arduino_ILI9486::writeAddrWindow(int16_t x, int16_t y, uint16_t w, uint16_t h)
{
  if ((x != _currentX) || (w != _currentW))
  {
    _currentX = x;
    _currentW = w;
    x += _xStart;
    _bus->writeC8D16D16(ILI9486_CASET, x, x + w - 1);
  }
  if ((y != _currentY) || (h != _currentH))
  {
    _currentY = y;
    _currentH = h;
    y += _yStart;
    _bus->writeC8D16D16(ILI9486_PASET, y, y + h - 1);
  }

  _bus->writeCommand(ILI9486_RAMWR); // write to RAM
}

void Arduino_ILI9486::invertDisplay(bool i)
{
  _bus->sendCommand((_ips ^ i) ? ILI9486_INVON : ILI9486_INVOFF);
}

void Arduino_ILI9486::displayOn(void)
{
  _bus->sendCommand(ILI9486_SLPOUT);
  delay(ILI9486_SLPOUT_DELAY);
}

void Arduino_ILI9486::displayOff(void)
{
  _bus->sendCommand(ILI9486_SLPIN);
  delay(ILI9486_SLPIN_DELAY);
}

// Companion code to the above tables.  Reads and issues
// a series of LCD commands stored in PROGMEM byte array.
void Arduino_ILI9486::tftInit()
{
  if (_rst != GFX_NOT_DEFINED)
  {
    pinMode(_rst, OUTPUT);
    digitalWrite(_rst, HIGH);
    delay(100);
    digitalWrite(_rst, LOW);
    delay(ILI9486_RST_DELAY);
    digitalWrite(_rst, HIGH);
    delay(ILI9486_RST_DELAY);
  }

  _bus->batchOperation(ili9486_init_operations, sizeof(ili9486_init_operations));

  if (_ips)
  {
    _bus->sendCommand(ILI9486_INVON);
  }
  else
  {
    _bus->sendCommand(ILI9486_INVOFF);
  }
}
```

moononournation, 2022-11-14:
Please get update and check again. If found the rotation is mirrored, try rotation + 4.

amirjak0, 2022-11-14:
### **I Re-Tested all four directions**https://github.com/moononournation/Arduino_GFX/issues/172#issuecomment-1313100572 It all worked well  

### **Sorry it took so long ,Thank you very much for the help.**
### **This is the result  With my __ILI9486_MAR3502(have no touch screen) 3.5"**

[image]

ojaksch, 2022-11-14:
Great! Thanks for reporting back and confirming my fixes. @moononournation has already merged my Pull Request and expanded the rotation possibilities.
I think this issue can be closed now. Thanks to all participants.

amirjak0, 2022-11-14:
Thanks to you

closed 2022-11-25 by moononournation