moononournation/Arduino_GFX #182: ILI9488 wrong init

Pepepotero, 2022-07-19:
Hello
I have a TFT ILI9488 I have connected it to an ESP32 TTGO T8 1.7 as follows:
TFT    ESP32 PIN
SCK    25
B/C    32
MOSI 33
RST   EN (rst) 
I have loaded the clock.ino example by changing the lines: 
Arduino_DataBus *bus = new Arduino_ESP32SPI(32 /* DC */, -1 /* CS */, 25 /* SCK */, 33 /* MOSI */, -1 /* MISO */); 
and
Arduino_GFX *gfx = new Arduino_ILI9488_18bit(bus, 0, 0 /* rotation */, false /* IPS */);
Just after uploading the sketch, white screen. I press the reset button of the ESP32, white screen, I press again, IT WORKS. I press reset again, white screen, again, again, again.....
That is, it works randomly. Thinking it could be the cables, I changed them several times with the same result. I changed the TTGO for another and the same result, so thinking that the TFT could be defective, I bought another one, and got the same results.
If I change the RST pin to -1 when defining the gfx, just white screen. And if I change the pin both physically and, on the line, just white screen...
Therefore, it may be a bug in the library. Looking at the code I see that when setting the pin to != -1 it does a soft reset (as you can see there is a 0) but it doesn't come out correctly. I have looked at the datasheet on the screen and it seems that everything is correct in the library. I don't know what to look at anymore.
Is it a library bug or am I doing something wrong?
Thank you very much in advance.

moononournation, 2022-07-19:
I am testing with a makerfabs test kit, here are the link:
https://www.makerfabs.com/esp32-3.5-inch-tft-touch-capacitive-with-camera.html
and here is the testing video:
https://youtu.be/NkE-LhtLHBQ

If you cannot turn it on, most likely it is not ILI9488. you may try ST7796, ILI9486 or ILI9481 driver instead.

closed 2022-07-22 by moononournation

JohnTaipei, 2022-07-23:
please try
Arduino_GFX gfx = new Arduino_ILI9488_18bit(bus, -1, 0 / rotation /, false / IPS */);

your rst is EN not GPIO0

moononournation, 2022-07-23:
You mean you have proved ILI9488 does not work with RST connect to EN? Then don't do it.

JohnTaipei, 2022-07-24:
He said that he connected RST to EN pin
so rst set -1

RST have to connect to EN pin

moononournation, 2022-07-24:
then you should test with his example code first

JohnTaipei, 2022-07-25:
i did
it works well
https://www.youtube.com/shorts/VYESfyJWG50

ili9488 ,st7796 ,nt35510
RST connect to EN 
rst -->-1
work ok