moononournation/Arduino_GFX #183: Unknown operation id with esp32 and st7735 displays

XanCraft21, 2022-07-26:
Hello. I really love this library for it’s speed and capabilities, but there is a small problem. When using esp32, if i use any of the st7735 display drivers, something strange happens. If i open the serial monitor and reset the board, after the boot info will say “unknown operation id at” then some numbers. This also affects the display somehow, making it seem brighter and it doesn’t work correctly, like if i keep the screen all black and have an animated image, sometimes there will be a ghost image left behind, it’s very faint and is hard to get rid of unless i let it rest for a while. If you can investigate this problem and find a solution, i will really appreciate that. Thank you.

moononournation, 2022-07-26:
seems it is caused by read flash data corruption. may be lower the flash speed or provide a better power source.

XanCraft21, 2022-07-27:
It only happens for st7735, other drivers and pretty much every library i use work fine. I am giving it the required 5v usb supply and i’ve never had problems with the flash speed before. I will try lowering the speed and see what happens, but i think there might be a hard to spot bug in the code with an easy fix.

moononournation, 2022-07-27:
if only happen in ST7735, what is the SPI speed?

XanCraft21, 2022-07-27:
The absolute maximum i can make the display run without malfunction, 40mhz (actually 39999999 because Arduino is weird). Even with this problem the display seems to run ok with an unusual brightness and occasional ghosting from power down. I believe the problem is only during the initialization of the display within the cpp file in the library, although every other function works fine.

moononournation, 2022-07-27:
I remember the max speed for ST7735 should be 27000000, you may test it again with this setting.

XanCraft21, 2022-07-27:
The current issue is still there. I lowered the spi speed and the errors in Serial are still there. I even tried lowering the flash speed on the esp32. Any other ideas you have?

moononournation, 2022-07-27:
I never seem the error message  “unknown operation id at” and google no result. So I don't know where the message come from...

XanCraft21, 2022-07-27:
That is very strange. I even tried a different controller and still got the same errors. I can only assume the latest version of the esp32 core has a compatibility issue with some libraries. Thank you for your help. I will try my best to work around these errors. I will leave this post open in case other members have had the same problem as i did.

moononournation, 2022-07-27:
is it any problems when you run PDQgraphicstest with ST7735?

XanCraft21, 2022-07-27:
Sadly yes. Like i said up above I believe the esp32 core has compatibly errors. And about the ghosting issue is best visible on an IPS display. It also happens with other controllers even though there are no error messages.

moononournation, 2022-07-27:
I am also using ST7735, I just checked with latest arduino-esp32 without any problems.[image]

XanCraft21, 2022-07-27:
Ok. I am not sure why i am the only one with this problem. I will wait and see what happens over time. Thank you once again for your help.

411956158, 2022-08-13:
Hello, I have encountered the same problem on the same equipment, and the fault is in this part.

_gfx->begin();
    gfx->fillScreen(BLACK);
    Serial.println("GFX BEGIN");
    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");_

moononournation, 2022-08-13:
whats mean you have same problem? you also see  “unknown operation id at” ?

411956158, 2022-08-13:
Yes, as shown in the figure below.(／_＼)[image]

XanCraft21, 2022-08-13:
I think i might know the reason to the unknown operation id bug. In the file st7735 . cpp inside the folder display, the init commands are different to the other displays, there is no full init command list and only a gamma adjustment. That adjustment might have been conflicting with a bug in the latest expressif esp32 arduino package. There is also another problem, because there are no init commands like the ones used in other displays like st7789 . cpp, the display is over saturated and seems brighter. Using the same display with a different library does not have this problem. There is also a faint ghosting with ips display with the st7735 driver, like the 0.96 inch 80x160 display. Even with these bugs i still love using this library since it has the most display chip support and good controller support.

411956158, 2022-08-14:
I made many attempts but failed to get the screen to display the content. The serial port can print the correct benchmark information, and I can observe that the screen has undergone a series of refreshes, but there is no display content.
Here is my configuration.

```
#define GFX_BL 39 
#define TFT_RST 42

Arduino_DataBus *bus = new Arduino_ESP32PAR8(41/*TFT_DC*/, 40/*TFT_CS*/, 9 /* WR */, 10 /* RD */, 0 /* D0 */, 1 /* D1 */, 2 /* D2 */, 3 /* D3 */, 4 /* D4 */, 5 /* D5 */, 6 /* D6 */, 7 /* D7 */);

Arduino_GFX *gfx = new Arduino_ST7735(bus, TFT_RST, 0 /* rotation */, false /* IPS */, 128 /* width */, 128 /* height */, 0 /* col offset 1 */, 32 /* row offset 1 */,true);
```

When I don't change the data bus, I only change the chip selection to any chip except ST7735, although it can't be displayed, the serial no longer outputs error messages.
Incidentally, I'm sure the mistake happened in this step. When I reduced the speed to 2000000, the error still existed.
```
gfx->begin();
```

moononournation, 2022-08-14:
> 

The mentioned code is related to #149 , may be I try to align this to display class

moononournation, 2022-08-14:
Please help to get update and check if the latest code can fix your problems.

411956158, 2022-08-14:
Thank you very much for your willingness to devote so much time and energy to my problem, but unfortunately, the mistake is the same as before.

411956158, 2022-08-14:
It is worth adding that I noticed that you seem to be using SPI mode in the photos you sent, while my screen is 8-8bit parallel.

411956158, 2022-08-16:
I just removed the gamma adjustment in the ST7735 file and the error disappeared. Although the program worked normally, it was strange that my screen didn't display correctly.

moononournation, 2022-08-17:
please check is it a latest code is work for you

XanCraft21, 2022-08-18:
So far for me it is still the same. The display colors are probably improved but the ips displays still have the faint ghosting and the unknown operation error is still there.

moononournation, 2022-08-18:
any pictures?

XanCraft21, 2022-08-18:
I do not have them ready at this time but i will take some soon. Thank you.

XanCraft21, 2022-08-22:
I have found another clue about this situation. Under arduino_databus . cpp in the function batch operation, it will print out the error “unknown operation is at” with some numbers if it finds an invalid code. Something in the st7735 drivers is affecting that function. Maybe if the gamma adjustment is removed and the initialization code from the tft_espi st7735 were used instead might fix the problem.

XanCraft21, 2022-08-29:
In my case the problem is still there, and i feel like it will never get fixed. I will soon close this issue. Thank you for the help you were able to give me.

XanCraft21, 2022-09-17:
I’m very sorry for being late to come back to this. I thought this was not going to get fixed, but i was wrong. With the newest update the unknown operation error has been fixed and the displays bugs i found are also fixed. Sorry for being impatient, thank you for your help.

closed 2022-09-17 by XanCraft21