moononournation/Arduino_GFX #184: Which databus should be selected for the 8bit parallel mode of ESP32S3?

411956158, 2022-08-01:
Dear developer, 
  thank you for your great works. He has benefited me a lot.
  I noticed that ESP32S3 i80 LCD parallel 16-bit has been supported. How can I drive an 8bit parallel LCD? Or it will take some time to be supported?
  I look forward to receiving your reply, pay my highest respect to you, and thank you again for your great promotion of LCD driver.

moononournation, 2022-08-01:
I am building driver when I need to use it. I just received an ESP32S3 dev device with 8-bit display, so it should support soon...

moononournation, 2022-08-01:
ESP32PAR8 or SWPAR8 can be used.

411956158, 2022-08-01:
> 

Thank you for your reply. I am very excited to hear this news and look forward to its early release.

SpaceTeddy, 2022-08-05:
also interresting for me. Currently I try to get bright a LCD with GC9A01 and 8-bit MCU interface. I'm quite sure, that my wirering is correct.

I uses as bus interface:
`Arduino_DataBus *bus = new Arduino_ESP32PAR8(13 /* DC */, 12 /* CS */, 14 /* WR */, GFX_NOT_DEFINED /* RD */, 27 /* D0 */, 26 /* D1 */, 25 /* D2 */, 33 /* D3 */, 32 /* D4 */, 15 /* D5 */, 2 /* D6 */, 4 /* D7 */);`

and for LCD:
`Arduino_GFX *gfx = new Arduino_GC9A01(bus, 5 /* RST */, 0 /* rotation */, true /* IPS */);`

any hints I can use or test? it is possible to read out data from the driver? till now, RD is not connected to the lcd

thx
Chris

SpaceTeddy, 2022-08-08:
jfi,
with my nodemcu32 the lcd was not able to light it up with bus driver 
`Arduino_DataBus *bus = new Arduino_ESP32PAR8(13 /* DC */, 12 /* CS */, 14 /* WR */, 2 /* RD */, 27 /* D0 */, 26 /* D1 */, 25 /* D2 */, 33 /* D3 */, 32 /* D4 */, 15 /* D5 */, 21 /* D6 */, 4 /* D7 */);`

I had success with:
`Arduino_DataBus *bus = new Arduino_SWPAR8(13 /* DC */, 12 /* CS */, 14 /* WR */, 2 /* RD */, 27 /* D0 */, 26 /* D1 */, 25 /* D2 */, 33 /* D3 */, 32 /* D4 */, 15 /* D5 */, 21 /* D6 */, 4 /* D7 */);`

it works fine now.

moononournation, 2022-08-17:
In my previous experience, Arduino_ESP32PAR8 may be too fast for some display. So same setting Arduino_SWPAR8 can run well in lower speed. Now in ESP32S3 you can use the Arduino_ESP32LCD8, it can adjust speed while init, e.g.:
```
gfx->begin(8000000);
```

closed 2022-09-10 by moononournation