moononournation/Arduino_GFX #188: SPI speed selection by user

ejdp62, 2022-08-06:
Hello, in my system I have an ST7735 greentab display, which is working fine with SWSPI from your library.
I use the HelloWorld sketch to see if the library works.

I would like to start using HWSPI, but this must be done at an SPI speed that is lower than what the library currently chooses,
most probably due to the long bus wires. (Unfortunately, I can't make them shorter easily...)

How do I set up a lower SPI speed from user code? (I would not like to hack the library itself)

moononournation, 2022-08-06:
gfx->begin(8000000); // 8MHz

ejdp62, 2022-08-08:
Hello, I followed your instructions (good to see that it is so simple!), but unfortunately the screen stayed blank.
To check if perhaps I had done something wrong, I hooked up my scope, and checked the serial clock and data lines.
 I can confirm by measurement that the serial clock behaves correctly, and is set according to the speed parameter.
The serial data line also shows expected behavior.

So now i have the following curious situation:

-The display works with the Adafruit library with SWSPI
-The display works with the Adafruit library with HWSPI (library must be hacked to lower the speed)
-The display works with the Arduino_GFX library with SWSPI
-The display does not work with the Arduino_GFX library with HWSPI (even if speed is set lower than for Adafruit library)

I have edited the HelloWorld example sketch to test functionality. I have only changed the global section
and the setup function, the loop function has not been changed. The code I have now is as follows:

============================================
#include <Arduino_GFX_Library.h>

#define SPI_PIN_DC   8
#define SPI_PIN_CS   GFX_NOT_DEFINED
#define SPI_PIN_SCK  13
#define SPI_PIN_MOSI 11
#define SPI_PIN_MISO GFX_NOT_DEFINED
#define SPI_SPEED    1000000

#define GFX_PIN_RST  9
#define GFX_ROT      0
#define GFX_IPS      false
#define GFX_WIDTH    128
#define GFX_HEIGHT   160
#define GFX_COL_OFS1 2
#define GFX_ROW_OFS1 1
#define GFX_COL_OFS2 2
#define GFX_ROW_OFS2 1

Arduino_DataBus *bus = 
new Arduino_HWSPI(SPI_PIN_DC,SPI_PIN_CS);
//new Arduino_SWSPI(SPI_PIN_DC,SPI_PIN_CS,SPI_PIN_SCK,SPI_PIN_MOSI,SPI_PIN_MISO);

Arduino_GFX *gfx = new Arduino_ST7735(bus,GFX_PIN_RST,GFX_ROT,GFX_IPS,
  GFX_WIDTH,GFX_HEIGHT,GFX_COL_OFS1,GFX_ROW_OFS1,GFX_COL_OFS2,GFX_ROW_OFS2);

/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
    gfx->begin(SPI_SPEED);
    gfx->fillScreen(BLACK);

    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    delay(5000); // 5 seconds
}

======================================================
This code does not work, but when in the bus declaration the SWSPI is chosen, the display works.

So, it seems my SPI hardware is OK (speed reduction is needed because of the long bus lines),
all pins are properly connected, etc. How would you suggest to proceed with debugging this?

P.S. We have now entered a situation where the original title of this issue does not apply anymore.
Should a new issue be opened to further process this?

closed 2022-08-10 by ejdp62