moononournation/Arduino_GFX #189: GC9503V on ZX3D95CE01S_AR question

poky, 2022-08-06:
Hi, I'm trying to get this display to run, so I used the example code here:
https://github.com/moononournation/Arduino_GFX/blob/87193ecf7acf4ad093ee1ab2b20431db0d4d10f8/examples/PDQgraphicstest/PDQgraphicstest.ino#L140

Unfortunately, it's non-runner, so I've found the sch, the pins are different, could anyone enlightens me how to config these pins? Thanks!

https://doc.smartpanle.cn/_static/espboard/ESP32S3/ZX3D95CE01S-AR/hardware.pdf

[image]

moononournation, 2022-08-14:
Yes the schematic in your link not the same as mind, can you drive it after change the pins?
[image]

poky, 2022-08-15:
> Yes the schematic in your link not the same as mind, can you drive it after change the pins?
> [image]

Hi, I've made the changes for the new schematic

```
#elif defined(ZX3D95CE01S_AR)
#define TFT_BL 4
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    0 /* CS */, 45 /* SCK */, 48 /* SDA */,
    40 /* DE */, 41 /* VSYNC */, 42 /* HSYNC */, 39 /* PCLK */,
    18 /* R0 */, 8 /* R1 */, 3 /* R2 */, 46 /* R3 */, 9 /* R4 */,
    13 /* G0 */, 12 /* G1 */, 11 /* G2 */, 10 /* G3 */, 16 /* G4 */, 17 /* G5 */,
    45 /* B0 */, 48 /* B1 */, 47 /* B2 */, 21 /* B3 */, 14 /* B4 */
);
```

I can see the backlit, but nothing shown on the screen, the console does have outputs:
```
Pixels	531287
Lines	671598
Horiz/Vert Lines	82890
Rectangles (filled)	517181
Rectangles (outline)	68685
Triangles (filled)	225237
Triangles (outline)	49928
Circles (filled)	48884
Circles (outline)	58404
Arcs (filled)	47631
Arcs (outline)	149474
Rounded rects (filled)	553896
Rounded rects (outline)	73701
Done!
```

Could you point out what might be wrong? Thanks!

moononournation, 2022-08-15:
may be your display is ST7701

moononournation, 2022-08-15:
you may try type1, type2 and type3

poky, 2022-08-15:
> you may try type1, type2 and type3

Do I just change thses:

```
Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
  bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */,
  true /* IPS */, 480 /* width */, 480 /* height */,
  st7701_type1_init_operations, sizeof(st7701_type1_init_operations),
  true /* BGR */);
```

To:

```
Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
  bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */,
  true /* IPS */, 480 /* width */, 480 /* height */,
  st7701_type2_init_operations, sizeof(st7701_type2_init_operations),
  true /* BGR */);
```

Because I have tried 1-3, but still the same, thanks!

Myzhazha, 2022-09-07:
Hi, I met the same problem. Have you solved it?

moononournation, 2022-09-07:
there are a pull request fixed type3 init code previously, you may try latest type3 init code again.

moononournation, 2022-09-10:
If you are using ZX3D95CE01S_AR, you can test by uncomment PDQgraphicstest.ino at around line 29:
```
#define ZX3D95CE01S_AR
```

closed 2022-09-10 by moononournation