moononournation/Arduino_GFX #190: HWSPI not functional for UNO and ST7735 GREENTAB display

ejdp62, 2022-08-10:
Hello, 

I am trying to get your Arduino_GFX library up and running with HWSPI on my UNO plus ST7735 greentab display.
Everything works fine with SWSPI. I do have a speed issue on my SPI bus, so I must lower the speed to get good results.
You have given me some advice how to do that in issue #188 (thank you!), and I have seen on my scope 
that throttling the SPI speed works correctly, by probing the serial clock and data lines.

After some experimentation, I now i have the following remarkable situation:

-The display works with the Adafruit library with SWSPI
-The display works with the Adafruit library with HWSPI (library must be hacked to lower the speed)
-The display works with the Arduino_GFX library with SWSPI
-The display does not work with the Arduino_GFX library with HWSPI (even if speed is set lower than for Adafruit library)

I have edited the HelloWorld example sketch to test functionality. I have only changed the global section
and the setup function, the loop function has not been changed. The code I have now is attached.
The attached code does not work, but when in the bus declaration the SWSPI line is chosen, the display works.
[HelloWorld.txt](https://github.com/moononournation/Arduino_GFX/files/9303005/HelloWorld.txt)

So, it seems my SPI hardware is OK (speed reduction is needed because of the long bus lines),
all pins are properly connected, etc. How would you suggest to proceed with debugging this?

P.S. The reason that the Adafruit library must be hacked to lower the speed, is because the initialization
of the display can not be slowed down. The speed reduction is applied _after_ initialization.
Could it be that the same is true for your library?

moononournation, 2022-08-10:
seems the long wires are the main cause. if SWSPI can work, keep using it.

ejdp62, 2022-08-11:
>seems the long wires are the main cause.

The display works with HWSPI at 4 MHz when I use the (hacked) Adafruit library.
That means the hardware is ok, just as long as the speed is set appropriately.
The Arduino_GFX library does not work even with HWSPI speed set at 1 MHz.
(See the code attached to the previous post.)
So the conclusion is that the root cause is in the SW, and not the HW.

>if SWSPI can work, keep using it.

When I need to choose between the four possibilities mentioned in my previous post,
at this point the Adafruit HWSPI option gives me by far the highest speed.
In principle, I would like to use your library, but only if it brings more speed. 
For that, the HWSPI option would need to work.

moononournation, 2022-08-11:
have you tried is HWSPI work without long wire?

ejdp62, 2022-08-13:
> have you tried is HWSPI work without long wire?

Unfortunately, my setup does not allow to make the wires shorter.
The fact that HWSPI can be made to work by lowering the speed tells me that the HW itself is working just fine.

The question is: If Adafruit HWSPI works at 4 MHz, then why won't Arduino_GFX HWSPI work even at 1 MHz?

moononournation, 2022-08-13:
in most case long wire and high speed cannot live together

ejdp62, 2022-08-13:
> in most case long wire and high speed cannot live together

Yes, that is quite correct! And it is precisely why I need to lower the HWSPI speed to 4 MHz.

When hacking the Adafruit library to set a speed of 4 MHz, the display works correctly with HWSPI.
So, my hardware setup (uno, SPI bus, display) works correctly using HWSPI at 4 MHz.

Therefore, 4 MHz is low enough for my SPI bus and display to work correctly.
That means that the Arduino_GFX library should also work correctly with HWSPI at 4 MHz,
because the hardware is precisely the same. 

What is the difference between bus signalling or commands sent between SWSPI and HWSPI?

moononournation, 2022-08-13:
Arduino UNO normally running SPI at 4MHz, so it is not lower the speed. The ST7735 works fine at 4MHz in my side. I cannot fix the problem that cannot reproduce in my side. In contrast, if find how to fix it, welcome for your contribution.

ejdp62, 2022-08-13:
> Arduino UNO normally running SPI at 4MHz

I have an Arduino Uno R3, running at 16 MHz, and the SPI clock divider is by default set at 2, 
resulting in a serial clock of 8 MHz. I have actually seen this speed on my oscilloscope!

> so it is not lower the speed.

When I specify 8 MHz, the HWSPI actually runs at 8 MHz, 
both with the Adafruit and also with the Arduino_GFX library.
So I do need to lower the default speed of 8 MHz down to 4 MHz.

> The ST7735 works fine at 4MHz in my side.

I have no doubt about that!

> I cannot fix the problem that cannot reproduce in my side.

I fully agree. This is why is asked in my very first post:

> How would you suggest to proceed with debugging this?

You know the ins and outs of the Arduino_GFX library, 
and perhaps you have some idea where I may look.

For example, in the bus code, i see that there is code to make a delay:
=======
    case DELAY:
      delay(batch[++i]);
      break;
=========

I do not know when how these delays are used, but if the _required_ length somehow depends 
on the rate of the serial clock, it could be worthwhile to try different values. 
One thing I do not know is if the delay can (in principle at least) be made arbitrarily long, 
or if there is some upper bound that must be adhered to.

Undoubtedly, there are other possibilities like this one, and you may have ideas of what I could try.
So please let me know how best to proceed in finding the root cause of this.
I will do my best to follow your suggestions!

> In contrast, if find how to fix it, welcome for your contribution.

Indeed, it is likely that there are other people who have a setup with similar limitations.
So whatever I find I will share with you, so you can incorporate it into the library,
which will benefit those other users.

ejdp62, 2022-08-13:
oops, something went wrong with the formatting....
My apologies, I did not mean to insert such big text!

moononournation, 2022-08-13:
Thanks for your contribution, what is the signal different you found in oscilloscope between adafruit and this library?

ejdp62, 2022-08-13:
On my scope, I can see no significant difference in the wave forms between the two libraries.
In both cases, the serial clock is a relatively nice square wave of the appropriate frequency when active. 
I think we would need to look at the commands and data that are sent to the display.

Something goes wrong in the initialization phase of the display, so at the very beginning of the sketch.
Could it be that similar to the not-hacked Adafruit library, the speed setting is applied too late?
The tricky thing of such a bug is that it is invisible on "normal" systems where SPI speeds of 8 MHz are allowed.
On such systems a bug like that goes unnoticed, even if some lower speed setting is used.

moononournation, 2022-08-14:
I have just fixed a bug #183 related to ST7735, please get update and check is it also can fix your problems as well.

ejdp62, 2022-08-14:
ok, sure! I just checked with the arduino library manager, and that one is not aware of the update yet.
So I will download the github code.

moononournation, 2022-08-14:
not yet release new version before confirm bugfix

ejdp62, 2022-08-14:
Hello, I just checked the latest version that you checked in 3 hours ago.
Downloaded the code, and replaced the library by hand, to make sure that it has been replaced properly.
Ran the sketch as in the first post, and the screen was blank. 
Changed the code to SWSPI (select the other bus declaration), and it worked.
So this fix does not solve my problem, unfortunately.

But I do have an important observation from this exercise! 
I can see a difference between the HW and the SW operation in a LED on the uno board.
The LED is marked "L", so it is the one connected to pin 13, the SPI serial clock!
in SWSPI mode, the led is off, while in HWSPI mode, the LED is on (and blinking off regularly for a short while).

Could it be that pin 13 is somehow inverted in HWSPI mode??

moononournation, 2022-08-14:
I don't think you can use GPIO 13 while using HWSPI. can you change to use another free pin?

ejdp62, 2022-08-14:
> I don't think you can use GPIO 13 while using HWSPI.

That is the point: I am _not_ using GPIO13!!! Pin 13 is fully driven by the HWSPI, and nothing else.
It is a lucky coincidence that the LED is attached to a HWSPI pin, and even more so: to the serial clock.
So, it seems that the HWSPI code somehow inverts the serial clock, as compared to the SWSPI code....

moononournation, 2022-08-14:
LED blink while SPI operating is normal, but you said SWSPI LED not blink?

ejdp62, 2022-08-14:
The impression of the led is mainly determined by its state during the main delays in the sketch 
(between the various renderings of "HelloWorld"). In SWSPI mode, the SPI line is sometimes high
(whenever a transaction is going on), but apparently not long enough to be clearly visible.
In the HWSPI case, the line is apparently high most of the time (between the various renderings of "HelloWorld"),
and sometimes it is low (whenever a transaction is going on). In this case, the turning off is indeed visible,
and the LED blinks.

It seems there are some bits not properly initialized. From the official Arduino documentation:
------------------------------------------------------------------------------------------------------
The SPI control register (SPCR) has 8 bits, each of which control a particular SPI setting.

| 7    | 6    | 5    | 4    | 3    | 2    | 1    | 0    |
| SPIE | SPE  | DORD | MSTR | CPOL | CPHA | SPR1 | SPR0 |

SPIE - Enables the SPI interrupt when 1
SPE - Enables the SPI when 1
DORD - Sends data least Significant Bit First when 1, most Significant Bit first when 0
MSTR - Sets the Arduino in controller mode when 1, peripheral mode when 0
CPOL - Sets the data clock to be idle when high if set to 1, idle when low if set to 0
CPHA - Samples data on the falling edge of the data clock when 1, rising edge when 0
SPR1 and SPR0 - Sets the SPI speed, 00 is fastest (4MHz) 11 is slowest (250KHz)

-------------------------------------------------------------------------------------------------------
The LED behavior indicates that the CPOL bit is not always set correctly in HWSPI mode.
And perhaps other bits as well, such as the CPHA bit!

ejdp62, 2022-08-14:
Hmmm.... It happened again.... Sorry about the large text! Can you repair that perhaps?

moononournation, 2022-08-14:
The underlying of the HWSPI is Arduino SPI class, normally I don't expect it will go wrong. And can you double check the SWSPI pins is the same as hardware SPI? Do you have photo about your connections?

ejdp62, 2022-08-14:
> And can you double check the SWSPI pins is the same as hardware SPI? Do you have photo about your connections?

It is even easier than that: you can see in the code that I attached to the original mail the pin assignments,
both for HWSPI and SWSPI. Here is the important snippet:

#include <Arduino_GFX_Library.h>

#define SPI_PIN_DC   8
#define SPI_PIN_CS   GFX_NOT_DEFINED
#define SPI_PIN_SCK  13
#define SPI_PIN_MOSI 11
#define SPI_PIN_MISO GFX_NOT_DEFINED
#define SPI_SPEED    1000000

#define GFX_PIN_RST  9
#define GFX_ROT      0
#define GFX_IPS      false
#define GFX_WIDTH    128
#define GFX_HEIGHT   160
#define GFX_COL_OFS1 2
#define GFX_ROW_OFS1 1
#define GFX_COL_OFS2 2
#define GFX_ROW_OFS2 1

Arduino_DataBus *bus = 
new Arduino_HWSPI(SPI_PIN_DC,SPI_PIN_CS);
//new Arduino_SWSPI(SPI_PIN_DC,SPI_PIN_CS,SPI_PIN_SCK,SPI_PIN_MOSI,SPI_PIN_MISO);

Arduino_GFX *gfx = new Arduino_ST7735(bus,GFX_PIN_RST,GFX_ROT,GFX_IPS,
  GFX_WIDTH,GFX_HEIGHT,GFX_COL_OFS1,GFX_ROW_OFS1,GFX_COL_OFS2,GFX_ROW_OFS2);

So, in both SWSPI and HWSPI cases, the serial clock is pin 13, and the serial data is pin 11.
I do not need to rewire the display when switching from HWSPI to SWSPI and back!

ejdp62, 2022-08-14:
Hello, I have been looking at the code in file Arduino_HWSPI.cpp as in the latest release of today.
Searching for and marking all occurrences of mySPISettings is illuminating.

Lines 7 through 17 deal with defining how to handle transactions.
Please note that line 8 is the *only* place where mySPISettings is used. 
It is defined in many places, but used only in line 8. So it is useful only when transactions are available. 

Line 20 is the static declaration of mySPISettings.

Then there are four places where mySPISettigns is defined, and it is in the final part of the begin() method.
The relevant lines are 148 through 201.

In lines 154, 161, and 169, mySPISettings are defined for a few architectures.
I assume that for these architectures transactions are indeed defined, otherwise the code would make no sense.
Also note that _spi->begin() is called in these three cases, and that the default data mode is MODE0.

Then, in lines 170 through 176, no test on architecture is made, only whether transactions are available.
_spi->begin() is called here as well, but now the default data mode is MODE2.

*I think line 174 contains the bug, and should use MODE0.*

The remainder of the section (lines 177 through 201) contains code that I do not completely understand, 
but it is clear that some speed settings are done that ignore any user-supplied information,
and there is a case that does not even call _spi->begin(). Some cleaning up might be useful there too.

ejdp62, 2022-08-14:
Update: I hacked line 174 as indicated, and now the HWSPI mode works!!!

ejdp62, 2022-08-14:
Please note that the final cases (lines 177 through 201) do seem to need some serious attention!

ejdp62, 2022-08-14:
Please note as well that once line 174 is corrected, the previous TWO cases (not three!) 
become redundant and can be omitted (under the condition that my previously mentioned assumption is true!)
The very first case (lines 148 through 154) has different code, and can not be omitted.

moononournation, 2022-08-14:
Thx for located ST7735 should use MODE0 in AVR, please get update and check is it solved.

ejdp62, 2022-08-15:
Hello, I downloaded the latest code directly from GitHub, and moved it into my Arduino libraries location.

> Thx for located ST7735 should use MODE0 in AVR

This is incorrect. Please refer to my post yesterday:

> Then, in lines 170 through 176, no test on architecture is made, only whether transactions are available. 
> _spi->begin() is called here as well, but now the default data mode is MODE2.
> 
> _I think line 174 contains the bug, and should use MODE0._

That line 174 does contain the bug was shown by a simple hack.
On inspection of the latest Arduino_HWSPI.cpp file however, I saw that line 174 still specifies MODE2.
Is that your intention?

ejdp62, 2022-08-15:
For comleteness sake, I also tried to run the code, but I get a compiler error:

/home/ej/ej/output/arduino/libraries/Arduino_GFX/src/display/Arduino_ST7735.cpp: In member function 'virtual void Arduino_ST7735::begin(int32_t)':
/home/ej/ej/output/arduino/libraries/Arduino_GFX/src/display/Arduino_ST7735.cpp:30:24: error: 'SPI_MODE0' was not declared in this scope
   _override_datamode = SPI_MODE0; // always use SPI_MODE0
                        ^~~~~~~~~
exit status 1
Error compiling for board Arduino Uno.

moononournation, 2022-08-16:
How long your wires actually?

ejdp62, 2022-08-16:
I can estimate the total length, including the length of the tracks on the Uno PCB as well as the TFT pcb.
You have to keep in mind that also some connectors and associated wiring are in the path towards the display.
So, the total length between the atmel pins and ST7735 pins is about 12 cm (5 inch).
And the complete path consists of a concatenation of several pieces that connect via connectors.

ejdp62, 2022-08-17:
hello, I have downloaded the latest version of the Arduino_GFX library, and installed it on my system.
I use the modified HelloWorld sketch as attached to the first post in this thread to test the library.
Result: the code compiles without errors, and also executes correctly!!
(And indeed, the "L" led now behaves in the HWSPI mode the same as in the SWSPI mode.)

Very glad to see this issue solved. Looking back, the bug must have been active for *all* ST7735 users!

moononournation, 2022-08-17:
Should be all AVR user, previously checked ESP32 still can display even hit this bug.

moononournation, 2022-08-17:
good to see it is ok now

closed 2022-08-17 by moononournation

ejdp62, 2022-08-17:
There are some aspects regarding this issue the could use further clarification.
For instance, that the MODE settings are determined by the slave, not the master,
and what the role of transactions is.

The code in Arduino_HWSPI.cpp shows that these aspects are not handled correctly.
If you like, we can discuss these aspects, and how best to deal with them in the library.
Just let me know how you would like to discuss!

moononournation, 2022-08-17:
This is outside this issue scope, you can raise another issue pointing out it.