moononournation/Arduino_GFX #193: ESP32 + ILI9486 16-bit Parallel

aikboon2002, 2022-08-21:
Hi! 

Have you tried with this combination, ESP32 + ILI9486 16-bit Parallel LCD? 
I have a 30-pin ESP32 board and I used up all pins including TX0 and TX2.
I assume DC is equalvalent to RS on my LCD. 
http://www.lcdwiki.com/res/MRB3952/4.0inch_16BIT_Module_MRB3952_User_Manual_EN.pdf 

I am using the following codes for GFX init. 
I have only white screen when I tried to run my code. 
Is there any clues why it doesn't work? 
Thanks for your help. 

// Connect CS pin to GND keeps enable, RD connect to Vcc keeps disable.
Arduino_DataBus *bus = new Arduino_ESP32PAR16(32 /* DC */, GFX_NOT_DEFINED /* CS */, 21 /* WR */, GFX_NOT_DEFINED /* RD */,
                                               19 /* D0 */, 23 /* D1 */, 18 /* D2 */, 5 /* D3 */, 22 /* D4 */, 1 /* D5 */, 25 /* D6 */, 26 /* D7 */,
                                               27 /* D8 */, 14 /* D9 */, 12 /* D10 */, 13 /* D11 */, 15 /* D12 */, 2 /* D13 */, 17 /* D14 */, 4 /* D15 */);
Arduino_GFX *gfx = new Arduino_ILI9486(bus, 33 /* RST */, 1 /* rotation */, false /* IPS */);

aikboon2002, 2022-08-22:
@moononournation Do you think it'll be better if I send you the display for trying? I have a spare display sitting idle. Thanks!

moononournation, 2022-08-23:
any pictures?

aikboon2002, 2022-08-23:
Please ignore the blue line as it's the screen protector. 

[image]

moononournation, 2022-08-24:
Normally I only accept sample display from manufacturer. I have never seen this type of display before, where you buy it?

aikboon2002, 2022-08-24:
Bought from Taobao. 
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.75dd2e8d8dC6GM&id=528278026903&_u=o1hb8php28c0

closed 2022-09-01 by aikboon2002

moononournation, 2022-10-11:
I found some incorrect interpretation of function in Arduino_ESP32PAR16 and now fixed. Please get update in github and test again.

aikboon2002, 2022-10-11:
Thanks! Will try and feedback.