moononournation/Arduino_GFX #194: ESP32 SPI fail with another device

PM04290, 2022-08-21:
I use TFT 2.4" ST7789 with LoRa module Ra01 (library https://github.com/sandeepmistry/arduino-LoRa)

//--- ESP32 SPI pin ---
// SCK  : 18
// MISO : 19
// MOSI : 23

#define PIN_TFT_CS  17
#define PIN_TFT_RST 16
#define PIN_TFT_DC  15

Arduino_DataBus *bus1 = new Arduino_ESP32SPI(PIN_TFT_DC, PIN_TFT_CS, DF_GFX_SCK, DF_GFX_MOSI, GFX_NOT_DEFINED /* MISO */, VSPI /* spi_num */);
Arduino_DataBus *bus2 = new Arduino_HWSPI(PIN_TFT_DC, PIN_TFT_CS);

gfx run well with bus2 but not correctly with bus1, there is incomplete drawing for drawRect for exemple

PM04290, 2022-08-21:
another thing, if EEPROM.begin() is called after gfx.begin(), gfx is lost.

moononournation, 2022-08-21:
you need input miso pin(19) instead of GFX_NOT_DEFINED if you need share the SPI.

PM04290, 2022-08-21:
ok, thanks.
it's better to use ESP32SPI rather than HWSPI for ESP32 ? if yes, why ?

PS: I am amazed by your library... very easy to implement and very efficient, both in code size and execution speed. well done.

moononournation, 2022-08-21:
ESP32SPI support 9-bit data for certain display without DC pin. For normal SPI display, it is not much different.

closed 2022-08-24 by moononournation