moononournation/Arduino_GFX #195: How to config  TFT 480x272  (ST7701) connected to  ESP32S3 ?

TridentTD, 2022-08-24:
My TFT is 480x272 ST7701 type  with the following pin connected to ESP32S3 pins.

[image]

How to config to use this TFT?

Thank you.

moononournation, 2022-08-24:
you may refer to Wiki example:
https://github.com/moononournation/Arduino_GFX/wiki/Display-Class#st7701-ips-lcd-480x480

closed 2022-08-24 by moononournation

TridentTD, 2022-09-18:
Hi, as you advised , I have tried by following the Arduino_GFX's wiki link .

[image]

I configes as the following...

```c
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  21   /* CS */,            // <------------ Right?
  14  /* SCK */,            // <------------ Right?
  ???  /* SDA */,           // <------------ Which pin ?
  
  40 /* DE */, 48 /* VSYNC */, 47 /* HSYNC */,  
  ??? /* PCLK */,         // <------------ Which pin ?
  
  4 /* R0 */, 5 /* R1 */, 6 /* R2 */, 7 /* R3 */, 15 /* R4 */,
  16 /* G0 */, 17 /* G1 */, 18 /* G2 */, 8 /* G3 */, 3 /* G4 */, 46 /* G5 */,
  9 /* B0 */, 10 /* B1 */, 11 /* B2 */, 12 /* B3 */, 13 /* B4 */
);

```

However , I don't know  how to config    CS pin,  SCK pin, SDA pin and  PCLK pin.
Do you mind to guild how to config the pins ?  

Thank you.

moononournation, 2022-09-18:
https://github.com/moononournation/Arduino_GFX/wiki/Display-Class#raspberry-pi-dpi-display

ashishwadekar, 2023-09-02:
@TridentTD If you are still looking for the setup, it will go this way:
```
#include <Arduino_GFX_Library.h>
#define GFX_BL 45

Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, 1 /* CS */,
    12 /* SCK */, 11 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 48 /* VSYNC */, 47 /* HSYNC */, 14 /* PCLK */,
    4 /* R0 */, 5 /* R1 */, 6 /* R2 */, 7 /* R3 */, 15 /* R4 */,
    16 /* G0 */, 17 /* G1 */, 18 /* G2 */, 8 /* G3 */, 3 /* G4 */, 46 /* G5 */,
    9 /* B0 */, 10 /* B1 */, 11 /* B2 */, 12 /* B3 */, 13 /* B4 */,
    1 /* hsync_polarity */, 60 /* hsync_front_porch */, 10 /* hsync_pulse_width */, 80 /* hsync_back_porch */,
    1 /* vsync_polarity */, 60 /* vsync_front_porch */, 10 /* vsync_pulse_width */, 20 /* vsync_back_porch */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 272 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, st7701_type1_init_operations, sizeof(st7701_type1_init_operations));
```

Cheers!