moononournation/Arduino_GFX #203: how to use STM32PAR8 dataBus?

liyang5945, 2022-09-16:
I have a STM32H750 dev board and a 8bit 8080 LCD，I use Arduino_SWPAR8 dataBus like this ,it works，but  it's too slow
```
Arduino_DataBus *bus = new Arduino_SWPAR8(
        PE10,PE11,PE13,PE12,
        PA0,PA1,PA2,PA3,PA4,PA5,PA6,PA7
);
```
I try to use STM32PAR8 dataBus 
```
Arduino_STM32PAR8::Arduino_STM32PAR8(int8_t dc, int8_t cs, int8_t wr, int8_t rd, GPIO_TypeDef *port)
    : _dc(dc), _cs(cs), _wr(wr), _rd(rd), _port(port)
{
}
```

It makes me confused, I don't how to setup the data pin，could you give me some advice?

moononournation, 2022-09-16:
You are using GPIO port A pin 0-7 for the 8-bit data, so you can pass the PORTA to last parameter.

liyang5945, 2022-09-19:
I pass the last parameter with `PortA` or `PORTA`，the code can't compile success，I change the parameter to `GPIOA` or `GPIOPort[PortA]`, the code can compile success，but dosen't work,the screen got nothing,I have two 8bit LCD screen，ST7796 and HX8352B，both can't work

moononournation, 2022-09-19:
@monte-monte

monte-monte, 2022-09-19:
```
Arduino_DataBus *bus = new Arduino_STM32PAR8(PB7 /* DC */, PB4 /* CS */, PB6 /* WR */, PB0 /* RD */, GPIOA /* PORT */);
```
That's what I use in my sketch and it works.

liyang5945, 2022-09-20:
> ```
> Arduino_DataBus *bus = new Arduino_STM32PAR8(PB7 /* DC */, PB4 /* CS */, PB6 /* WR */, PB0 /* RD */, GPIOA /* PORT */);
> ```
> 
> That's what I use in my sketch and it works.

what dev board do you use?

monte-monte, 2022-09-22:
@liyang5945 bluepill

closed 2022-10-30 by moononournation