moononournation/Arduino_GFX #208: Colors are wrong with ST7789

TimoKettunen, 2022-09-24:
I have a MorphESP 240 that has a ST7789 display. The color constants that are predefined (like BLACK) do not produce correct color.
Following script demonstrates the problem

#include <Arduino_GFX_Library.h>  // Core graphics library by Adafruit
#include <SPI.h>
#define TFT_DC 14
#define TFT_RST 9
#define TFT_CS 10    // only for displays with CS pin
#define TFT_MOSI 11  // for hardware SPI data pin (all of available pins)
#define TFT_SCLK 12  // for hardware SPI sclk pin (all of available pins)

Arduino_DataBus *bus = new Arduino_ESP32SPI(TFT_DC /* DC */, TFT_CS /* CS */, TFT_SCLK /* SCK */, TFT_MOSI /* MOSI */, GFX_NOT_DEFINED /* MISO */, FSPI /* spi_num */);
Arduino_ST7789 *tft = new Arduino_ST7789(bus, TFT_RST);  

void setup(void) {
  Serial.begin(115200);
  tft->begin();  
  Serial.println("Initialized");
  delay(5000);
  tft->fillScreen(BLACK); // Produces white
  delay(5000);
  tft->fillScreen(RED); // Produces cyan
  delay(5000);
  tft->fillScreen(BLUE); // Produces yellow
  delay(5000);
  tft->fillScreen(GREEN); // Produces magenta
  delay(5000);
  tft->fillScreen(YELLOW);// Produces blue
  delay(5000);
}

moononournation, 2022-09-25:
Below link are the video I post for the ST7789 graphics test previously, do you think the color is correct?
https://youtu.be/A961iGsJwA0

TimoKettunen, 2022-09-25:
That did not look like ST7789. The ST7789 is rectangular 240*240 display.
Anyway, could this be about color modes that the display supports? It has 12-, 16- and 18-bit modes. The constants are for 16-bit mode.

moononournation, 2022-09-25:
If you google ST7789, it has many dimension variations.

TimoKettunen, 2022-09-25:
OK. I did google but found only rectangular so I thought that is the only one.
When I first received this board, it had SW that showed correct colors. So HW is OK.
The I downloaded that [script](https://github.com/ccadic/ESP32-S2-DevBoardTFT/blob/master/code/EP32S2-display/EP32S2-display.ino) but I did not get that to compile, I did not find what GFX library it is using.
I tried Adafruit GFX but I got nothing displayed. The I tried yours and after some modifications (constructors, init -> begin etc.) it worked almost perfectly. It was significantly faster than the original SW (not a problem;) but the colors were off.
Is there anything in my example script that is wrong? It is dead simple and should work.

moononournation, 2022-09-25:
any video show whats going wrong?

TimoKettunen, 2022-09-25:
The comments in the script make it pretty obvious what goes wrong. If black is shown as white, what additional information the video can give? 
But here is one attached. I modified the script as below so that some text is also shown. The init done as in first post. 
```
  tft->setTextSize(2);
  tft->setTextColor(WHITE);
  tft->fillScreen(BLACK);
  tft->setCursor(0, 30);
  tft->println("This should be White text on BLACK");
  delay(1000);
  tft->fillScreen(RED);
  tft->println("This should be White text on RED");
  delay(1000);
  tft->fillScreen(BLUE);
  tft->println("This should be White text on BLUE");
  delay(1000);
  tft->fillScreen(GREEN);
  tft->println("This should be White text on GREEN");
  delay(1000);
  tft->fillScreen(YELLOW);
  tft->println("This should be White text on YELLOW");
```

https://user-images.githubusercontent.com/25174101/192156438-bb3eb90a-2df1-47d9-a12d-f73aeea34842.mp4

fvanroie, 2022-09-25:
The MorphESP 240 has an IPS Color Display. Try `tft->invertDisplay(true);` or set the IPS flag when instantiating Arduino_ST7789.

moononournation, 2022-09-25:
Sorry, I have missed the "do not produce correct color" means inverted color commented in the code. Yes, for the IPS version display, you can pass IPS parameter as true to fix(invert) the color.

moononournation, 2022-09-25:
For more parameter details, you can refer to PDQgraphicstest or WiKi.

TimoKettunen, 2022-09-26:
I now modified the script and used 'true' in constructor for ips. That fixed it!
The original script that I linked above did not have anything like that but it used different library, probably it handled it automatically.
Great thanks for help! Using this library for now.

closed 2022-09-29 by moononournation