moononournation/Arduino_GFX #215: ili9488 display error.  color direction error

haolsh, 2022-10-14:
```

#include <Arduino_GFX_Library.h>

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
Arduino_DataBus *bus = new Arduino_ESP32LCD16(
    45, 46, 42 /* WR */, 41 /* RD */,
    47 /* D0 */, 21 /* D1 */, 18 /* D2 */, 17 /* D3 */,16 /* D4 */, 15 /* D5 */, 14 /* D6 */, 13 /* D7 */,
    12 /* D8 */, 11 /* D9 */, 10 /* D10 */, 9 /* D11 */, 8 /* D12 */, 7 /* D13 */, 6 /* D14 */, 5 /* D15 */);
/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
Arduino_GFX *gfx = new Arduino_ILI9488(
  bus, 48 /* RST */, 2/* rotation */, false /* IPS */);
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/

void setup(void)
{
    Serial.begin(115200);
    gfx->begin();
    gfx->fillScreen(WHITE);
 
   // while (!Serial);
   Serial.printf("get screent width is %d and height is %d \n",gfx->width(),gfx->height());

    gfx->drawRect(0,0,100,100,RED);
    gfx->fillRect(0,0,100,100,RED);
   gfx->setTextColor(GREEN);
```

[image]

it doesn't show a 100*100 rect . 

but when i use LovyanGFX driver its display ok  like this
[image]

### what should i do ???  help!!!!!

moononournation, 2022-10-14:
try add a speed value while begin: gfx->begin(8000000);

haolsh, 2022-10-14:
> try add a speed value while begin: gfx->begin(8000000);

还没试，明天试下，大哥能解释下原理吗？频率为啥影响渲染方向😟😟，谢谢

haolsh, 2022-10-15:
> try add a speed value while begin: gfx->begin(8000000);

it doesnt work

moononournation, 2022-10-15:
may be related previous issue, please get latest v1.3.0 to test again.

haolsh, 2022-10-15:
> may be related previous issue, please get latest v1.3.0 to test again.

[image][image]

I defined rd pin to a gpio , but beginwrite function did not pull up rd pin. Has the influence?

moononournation, 2022-10-15:
it should pulled up at line 58

haolsh, 2022-10-17:
MCU:ESP32-S3  N8R8
SCREEN: ILI9488 3.5  320*480
ERROR: The x-axis color fill is wrong,it filled by y-axis . Change to LovyanGFX it display well .

try those:
1、try modify driver cmd  .
 result:not worked
2、get latest version .
result:not worked
3、modify begin speed.
result: not worked

close th issue .thanks,

closed 2022-10-17 by haolsh