moononournation/Arduino_GFX #217: Possibility to use Sprites?

petervanderwalt, 2022-10-31:
Currently when loading sprites/icons using drawbitmap function - you have to erase pixels first and redraw otherwise old position/old sprays still leaves some pixels behind: This creates a flashing onscreen 

https://github.com/m5stack/M5Stack/blob/master/src/utility/Sprite.h

moononournation, 2022-10-31:
Your code is using TFT_eSPI, I think you have post to a wrong place.

petervanderwalt, 2022-11-01:
No, its an example of a feature available for TFT_eSPI.  I don't use TFT_eSPI thus the feature request to add Sprite support to Arduino_GFX

moononournation, 2022-11-01:
What is your use case?

moononournation, 2022-11-02:
I think you can use this overload to do the job:
```C
void draw16bitRGBBitmap(int16_t x, int16_t y, uint16_t *bitmap, uint8_t *mask, int16_t w, int16_t h);
```

petervanderwalt, 2022-11-02:
Will give a test thanks

closed 2022-11-25 by moononournation