moononournation/Arduino_GFX #222: Compiling error - Arduino_STM32PAR8 + ILI9225-PAR8 + STM32F103C8T6 (Blue Pill)

ProtMonster, 2022-11-12:
I wanted to use the ILI9225 8-bit parallel display on my STM32F103C8T6 (Blue Pill), but I couldn't.

Use the following package and library:
STMF1xx/GD32F1xx boards, stm32duino (2021.05.31 [Y.m.d])
GFX Library for Arduino, Moon On Our Nation (1.3.0)

It is not possible to use it because ArduinoIDE gives an error when compiling:

Examples/Clock/Clock.ino
...
Arduino_DataBus *bus = new Arduino_STM32PAR8 (PB7 /* DC */, PB4 /* CS */, PB6 /* WR */, PB0 /* RD */, GPIOA /* PORT */);
Arduino_GFX *gfx = new Arduino_ILI9225(bus, PB9 /* RST */);
...

"
Clock.ino:36:28: error: expected type spec before 'Arduino_STM32PAR8'
  Arduino_DataBus *bus = new Arduino_STM32PAR8 (PB7 /* DC */, PB4 /* CS */, PB6 /* WR */, PB0 /* RD */, GPIOA /* PORT */);
                             ^
exit status 1
Compilation error: expected type-specifier before 'Arduino_STM32PAR8'
"

The "Arduino_SWPAR8" works with the same parameters.
What causes the error?

moononournation, 2022-11-13:
Sorry, I do not have blue pill in hand, so not yet support blue pill.

ProtMonster, 2022-11-13:
I see, thank you!

closed 2022-11-25 by moononournation