moononournation/Arduino_GFX #233: question about including #include "SPI.h" in display classes.

nielsnl68, 2022-12-11:
Hello,
I'm working on integrating your Library as a display component for esphome (www.esphome.io).

All go's almost okay, the only issue is that in some display drivers there is still the #include "SPI.h" in it.
Because of that my component will not compile right away.

Is there a special reason why this is in:
\GFX Library for Arduino\display\Arduino_GC9106.h
\GFX Library for Arduino\display\Arduino_GC9107.h
\GFX Library for Arduino\display\Arduino_GC9A01.h

espHome uses platformIO to compile there sourches. And when i try to compile i get the following error:

```
/GFX Library for Arduino/src/databus/Arduino_HWSPI.cpp:5:
.piolibdeps/esphome-web-c0b554/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:8:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
```

my source files can be found: https://github.com/nielsnl68/esphome-components/tree/main/components/graphics

moononournation, 2022-12-11:
This library is target used by Arduino IDE only.

nielsnl68, 2022-12-12:
Thanks for pointing that out, after talking to the espHome maintainer, the SPI.h issue was a was easily fixed by loading the SPI library before yours.  Now it is loading as it should be. Although i need to test it out now. From what i have seen it uses a very small footprint so far.

Still wondering why the SPI is loaded in those display driver files i shown above.

moononournation, 2022-12-12:
because it is needed