moononournation/Arduino_GFX #234: Error in compiling WiFiPhotoFrame.ino

nodoubtman, 2022-12-13:
Hello.
In WiFiPhotoFrame 
in JpegFunc.h

i have the error on this function

static int jpegOpenHttpStream(WiFiClient *http_stream, int32_t dataSize, JPEG_DRAW_CALLBACK *jpegDrawCallback)
{
  return _jpeg.open(http_stream, dataSize, jpegCloseHttpStream, jpegReadHttpStream, jpegSeekHttpStream, jpegDrawCallback);
}

:
not matching function for call to 'JPEGDEC::open(WiFiClient*&, int32_t&, void(&)...

Thank you.
Marc.

moononournation, 2022-12-13:
Please ensure both JPEGDEC and Arduino_GFX are in latest version

nodoubtman, 2022-12-13:
> Please ensure both JPEGDEC and Arduino_GFX are in latest version

Done.
Same error:

no matching function for call to 'JPEGDEC::open(WiFiClient*&, int32_t&, void (&)(void*), int32_t (&)(JPEGFILE*, uint8_t*, int32_t), int32_t (&)(JPEGFILE*, int32_t), int (*&)(JPEGDRAW*))'

moononournation, 2022-12-13:
The JPEGDEC commit not yet release: https://github.com/bitbank2/JPEGDEC/commit/1c00e9035ceff5d45fb86330e9469ecd0622db43

You need use master version.

nodoubtman, 2022-12-13:
I copied the lines.
Working perfectly...
just what i needed :)
Thanks.
Marc.

moononournation, 2022-12-14:
Look forward @bitbank2 JPEGDEC next release.

closed 2022-12-14 by moononournation

bitbank2, 2022-12-14:
I'm working on a new release now and will hopefully get it out by the end of today. I addressed a problem with odd sized images (partial MCUs). The new JPEGDRAW structure will have 2 new variables (iWidthUsed, iHeightUsed) to show clipped edges.