moononournation/Arduino_GFX #240: What license is this distributed under?

Anutrix, 2022-12-20:
Is it MIT License?

nullvalue0, 2022-12-22:
Please add a license to this project

moononournation, 2022-12-22:
As you know, it related to many source. All of them have different licenses, only an expert can determine what licenses it can be. Do you know any experts in this scope?

nullvalue0, 2022-12-22:
I guess that depends: did you rewrite the code from scratch or borrow heavily from those other projects in some places?

moononournation, 2022-12-22:
heavily

nullvalue0, 2022-12-22:
Then one approach is to release your code under the most-restrictive license that you referenced. But as it stands now without a license, noone can legally use your library for anything.

moononournation, 2022-12-22:
I am base on open source code and this is also open source project, so using my code is welcome. I just concern I released with an not appropriate licence restricted the usage. In contrast, I cannot represent the original source owner release it freely.

nullvalue0, 2022-12-22:
Well, "open source" means a lot of different things to different people. Please realize that releasing without a license IS THE MOST RESTRICTIVE option because it implies you retain all rights. This means no one is allowed to download, use or change your code. 

I hope someone with more expertise in this area can chime in and help you choose a license. In the meantime, you could start by compiling a list of all your referenced libraries and their respective licenses.

moononournation, 2022-12-22:
yes, I just need an expert opinion.

Anutrix, 2022-12-23:
I would suggest MIT license for the full repo/project and specifically borrowed parts be their respective licenses.

moononournation, 2022-12-23:
Actually you just want me release it with MIT, right? But I don't know is it appropriate for all the code. It is already cannot break apart.

mathertel, 2023-08-20:
You write that you used the Adafruit GFX Library <https://github.com/adafruit/Adafruit-GFX-Library> as a starting point.
Then just continue using the BSD license that is attached to it.
Please read it - and add the requested reference and copyright notices and a local LICENSE file.

From <https://github.com/adafruit/Adafruit-GFX-Library/blob/master/license.txt>

closed 2023-08-22 by moononournation