moononournation/Arduino_GFX #241: using a big screen with ardruino_canvas and not enough memory.

nielsnl68, 2022-12-20:
Hello @moononournation,

My esphome integration of your library is going very well at the moment.
There is one issue i came grosse today,
while using the canvas class when the psram is not enabled.
It gives an error but it is still allowed to try to write data in to the none initialized buffer memory. Which esp32 does not like very well.

Maybe the `begin()` method could return a boolean value that tells if it is okay to continue or maybe an raise exception that we can catch? 

BTW. You use `Serial.Println(...)` to send an error, there is a change the serial port is not init. My suggestion is to modify it to use `log_e`

moononournation, 2022-12-20:
Return code in begin() is a good idea, I will consider about it. log_e is Arduino-esp32 specific function and will not consider.

nielsnl68, 2022-12-20:
> ... log_e is Arduino-esp32 specific function and will not consider.
When you are considering the return code  for begin() maybe the developer using could decide on how to report the error to his/here end-user. and can the `Serial.println();` line be omitted.

moononournation, 2023-01-09:
Now it will return false if begin() allocate failed. Please get update and check.

nielsnl68, 2023-01-09:
thanks for taking the time to add this most appricated.

Just looked trough the changes for i was wondering should we not `return false;` in the Arduino_Canvas::begin() methods? 
True at this moment Arduino_TFT::begin() returns always true atm but it could be that in the future someone create a driver that will return false.

About the  `Arduino_TFT::begin();` is there no reason to expect that the `_bus->begin()` never fail? When it could fail, then i think we should also change does begin() methods to allow to return a boolean telling of the function was successful or not. Do you not agree?

moononournation, 2023-01-09:
yes, that will be next step

closed 2023-01-13 by moononournation

nielsnl68, 2023-01-13:
cool thanks for your work.