moononournation/Arduino_GFX #244: mages of another screen in the background of the currently running screen.

vijayflashbulb, 2022-12-23:
hi,
I'm working on a project which uses LVGL for graphics and to implement this I'm using this Arduino_GFX library.
I have designed multiple screens and the issue is sometimes I see images of another screen in the background of the currently running screen.
Does it have anything to do with this library.
[image]
the ring in the 1st image can be seen on the other screen(image below) also.
[image]

some details regarding the display are given below
driver = ILI9488
controller = esp32 s3

DataBus -> Arduino_DataBus* bus = new Arduino_SWPAR8(41 /* DC */, GFX_NOT_DEFINED /* CS */, 10 /* WR */, GFX_NOT_DEFINED /* RD */, 1 /* D0 */, 2 /* D1 */, 4 /* D2 */, 5 /* D3 */, 6 /* D4 */, 7 /* D5 */, 8 /* D6 */, 9 /* D7 */);

Arduino_GFX* gfx = new Arduino_ILI9488(bus, 42 /* RST */, 0 /* rotation */, true /* IPS */);

Thanks.

moononournation, 2022-12-23:
is it normal running LvglWidget example?

vijayflashbulb, 2022-12-23:
yes, i have the same issue.

moononournation, 2022-12-23:
is it better if using ESP32LCD8 instead of SWPAR8?

vijayflashbulb, 2022-12-23:
it is same.

moononournation, 2022-12-23:
any video on LvglWidget example?

vijayflashbulb, 2022-12-23:
[image]

when i restart and only turn on the backlit i can see some text and ring in the background
[image]
[image]

moononournation, 2022-12-23:
how about the operation of LvglWidget, is it display normal?

vijayflashbulb, 2022-12-23:
yes it works fine

vijayflashbulb, 2022-12-23:
can also see some data from the previous code sometimes.
i think it is a hardware issue.

moononournation, 2022-12-23:
May be related to the init code, have you tried other display libraries? Is it Similar?

vijayflashbulb, 2022-12-23:
no, i have not tried it yet.

radontec, 2022-12-26:
I´m facing the same problem and I asked the Chinese manufacturer.
I got this answer but I´m not sure what I can do with this information, maybe it helps you:

So you need to try and debug the code

LCD_CPU_WR_INDEX(0,0xB1); //Frame rate
LCD_CPU_WR_DATA(0,0xA0); //60Hz
LCD_CPU_WR_DATA(0,0x11);

LCD_CPU_WR_INDEX(0,0xB3);
LCD_CPU_WR_DATA(0,0x00);
LCD_CPU_WR_DATA(0,0x11);

moononournation, 2023-01-13:
Anyone tested the above init code changes?

closed 2023-03-04 by moononournation