moononournation/Arduino_GFX #247: can't get to work LCD SPI 2'' ST7789 320x240

ESPboy-edu, 2022-12-26:
lib is OK to work with ST7735 128x128, 128x160, ST7789 240x240
but can't get work this one LCD SPI 2'' ST7789 320x240

https://aliexpress.ru/item/1005004237117445.html?sku_id=12000028482573872&spm=a2g2w.productlist.search_results.12.2ae84aa63tc3rP
(it's not OLED as mentioned in description :) )

it works with TFT_eSPI without any problems
it has 5v logic and take into account that for 3.3v it is necessary to remove SMD voltage dividing resistors on the back side of the LCD

thanks for great work and brilliant gfx lib!

moononournation, 2022-12-26:
more details? I am using ST7789 in many projects without problems.

ESPboy-edu, 2022-12-28:
yes, your excellent lib works with different ST7789 in my projects too
but not with 2.0'' display mentioned above (but display is ok because it works with TFT_eSPI lib).
i'll do more tests recently and let you know

tapeboy7, 2023-01-18:
Make sure you have the reset pin defined and connected, It seems like the ST7789 chip is picky about that.
I had multiple GC9A01 displays working simultaneously, (they work fine without the RST pin connected) but I couldn't get a ST7789 based display work when I added it as a 3rd display, until I defined and connected the RST pin on it.