moononournation/Arduino_GFX #250: ESP8266 + ST7789  select SPI_MODE

hellvesper, 2023-01-03:
Hi! I'm using ESP8266 (Wemos D1 mini) + ST7789 240x240 tft display (without CS pin). And there was a bit tricky to make it work with this library. I was previously used this setup with Adafriut library, and It works well using `SPI_MODE2`:
```
Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
tft.init(240, 240, SPI_MODE2); // Init ST7789 240x240
```

But I don't know how to set spi mode without editing library code. Now I have workaround in the `Arduino_ST7789.cpp`:
```
void Arduino_ST7789::begin(int32_t speed)
{
#if defined(ESP32) || defined(ARDUINO_ARCH_NRF52840)
  _override_datamode = SPI_MODE3;
#endif
#if defined(ESP8266)
  _override_datamode = SPI_MODE2;
#endif
  Arduino_TFT::begin(speed);
}
```

But I'd like to provide needed arguments from setup section. How do I do this? I'm not very familiar with c/c++.

moononournation, 2023-01-04:
Ok, change applied. Please get update and test is it ok.

hellvesper, 2023-01-08:
> Ok, change applied. Please get update and test is it ok.

It works, thank you