moononournation/Arduino_GFX #257: ILI9331 display realoads on each canvas.begin()

peredozo, 2023-01-26:
Hello.

`
// ...

Arduino_GFX *disp = new Arduino_ILI9331( dispBus, 33, 0, false );

Arduino_GFX *cv1= new Arduino_Canvas(time_size[0], time_size[1], disp, time_size[2], time_size[3]);
Arduino_GFX *cv2 = new Arduino_Canvas(state_size[0], state_size[1], disp, state_pos[0][0], state_pos[0][1]);
Arduino_GFX *cv3 = new Arduino_Canvas(state_size[0], state_size[1], disp, state_pos[1][0], state_pos[1][1]);

// ...

void setup() {
  disp->begin(); // Display blinking ( ili9331_init_operations? )
  cv1->begin(); // Display blinking ( ili9331_init_operations? )
  cv2->begin(); // Display blinking ( ili9331_init_operations? )
  cv3->begin(); // Display blinking ( ili9331_init_operations? )
}

// ...
`

The display is reinitialized on each canvas initialization. I think that the initialization of the display should occur once (if i don't run begin() for each canvas - canvas doesn't work)

moononournation, 2023-01-26:
noted, I need more input for a better implementation and workable example

peredozo, 2023-01-28:
I'm not sure if this is the ideal solution, but it solves the problem of re-initializations.

Arduino_ILI9331.h
```
class Arduino_ILI9331 : public Arduino_TFT
{
...
private:
  ...
+  bool _FIRST_INITIALIZED = false;
};
```
Arduino_ILI9331.cpp
```
void Arduino_ILI9331::tftInit()
{
+	if(_ALREADY_INITIALIZED) return;
+	_ALREADY_INITIALIZED = true;
	...

```

h5n1xp, 2023-01-29:
It's my opinion that the begin() method doesn't make any sense on the canvas object and should be a NOP.

The Canvas class currently performs the buffer setup in the begin() method, but all this should be moved to the constructor (or a new method which allows more control over the buffer allocation).

moononournation, 2023-01-29:
As you may aware the Arduino IDE examples pattern, the constructor is outside setup().
So it is better not do the job in it.

h5n1xp, 2023-01-29:
> As you may aware the Arduino IDE examples pattern, the constructor is outside setup().
> So it is better not do the job in it.

That is a good point. Then the canvas needs a new buffer setup method. Perhaps overloading begin() so that by default it does the current initialisation but with the suggested guard variable, and add some overloads which allow more fine grained buffer allocation, i.e. different ram types/user defined/etc.

moononournation, 2023-02-23:
I have just added a constant `GFX_SKIP_OUTPUT_BEGIN` for skipping duplicate initialization, e.g.
```
canvas1.begin();
canvas2.begin(GFX_SKIP_OUTPUT_BEGIN);
canvas3.begin(GFX_SKIP_OUTPUT_BEGIN);
canvas4.begin(GFX_SKIP_OUTPUT_BEGIN);

```

closed 2023-03-04 by moononournation