moononournation/Arduino_GFX #258: Confusing performance figures

puijkh, 2023-01-27:
Could you please add to the performance figures what they mean. It is just numbers now. Do they indicate seconds or maybe a count of somewhat?

Thanks

moononournation, 2023-01-27:
what figure?

puijkh, 2023-01-27:
These ones:

Benchmark | Adafruit_GFX | Arduino_GFX | Lovyan_GFX | TFT_eSPI
-- | -- | -- | -- | --
Screen fill | 195,782 | 160,094 | 154,341 | 155,938
Text | 97,662 | 18,960 | 22,473 | 21,752
Pixels | 1,365,211 | 903,549 | 867,702 | 775,781
Lines | 1,062,311 | 412,026 | 269,060 | 264,950
Horiz/Vert Lines | 17,637 | 14,197 | 13,692 | 13,833
Rectangles-filled | 406,817 | 332,696 | 320,761 | 323,908
Rectangles | 11,641 | 9,254 | 8,545 | 8,714
Triangles-filled | 150,941 | 118,010 | 105,661 | 109,675
Triangles | 58,843 | 23,570 | 15,884 | 16,277
Circles-filled | 76,739 | 52,170 | 42,787 | 45,827
Circles | 118,125 | 40,955 | 25,959 | 25,269
Arcs-filled | N/A | 33,381 | 21,546 | N/A
Arcs | N/A | 66,054 | 47,901 | N/A
Rounded rects-fill | 408,534 | 338,136 | 318,882 | 323,189
Rounded rects | 43,185 | 21,562 | 13,089 | 15,371

moononournation, 2023-01-27:
those are milliseconds

closed 2023-02-17 by moononournation