moononournation/Arduino_GFX #260: Arduino_Canvas_Indexed not working as expected

h5n1xp, 2023-02-09:
I've been trying to improve the performance of my "Sprite" class, and one approach I have tried is to use 8bit GFX rather than 16bit, as the image data is stored in flash and the various buffers tend to be in ps_ram, the bottleneck could be the SPI interface of these devices so reducing the memory transactions should help.

It seems that the Arduino_Canvas_Indexed class actually builds the palette when colours are requested in real-time... Since the original 16bit colour values are required for this, the process is much slower and just as memory intensive (for my application) as the 16bit version.

My expected behaviour would be that during the initialisation of the Arduino_Canvas_Indexed class, one would specify a 256 colour array to use in the lookup.

Perhaps we need a new canvas class:

**Arduino_Canvas_Palette** 

Where the colour index can be predefined at initialisation, and not built in real-time, making it more suitable for games. This model is closer to the Atari ST and Commodore Amiga GFX approach to colour space.

h5n1xp, 2023-02-14:
I can work around the palette issue by looping through the palette calling get_index_colour() for each colour I need in the order I need them in the index, before calling any graphics methods.

I have now discovered that 8bit reads from ps_ram (and flash) are significant slower than 32bit reads

moononournation, 2023-02-14:
I think I should create another class called Arduino_Canvas_paletted, and the palette is explicit assigned. Then all GFX operation direct used color index.

h5n1xp, 2023-02-16:
> I think I should create another class called Arduino_Canvas_paletted, and the palette is explicit assigned. Then all GFX operation direct used color index.

Perhaps overload the drawing methods in the Arduino_Canavs_Indexed class so that a uint8_t colour value draws directly from the colour index, and a uint16_t value behaves the current way (creating a new colour entry if needed).

moononournation, 2023-02-17:
Added 2 new APIs in Arduino_Canvas_Indexed class:
```
  uint16_t *getColorIndex();
  void setDirectUseColorIndex(bool isEnable);
```

Please get update and help to check is it good for your work.

h5n1xp, 2023-02-17:
Many thanks!

closed 2023-03-04 by moononournation