moononournation/Arduino_GFX #262: Release 1.3.2 - Makerfabs 800x480 TFT with touch

ftensi, 2023-02-23:
The library work with the release 1.3.0. But with release 1.3.2 i have this error 
C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino:18:1: error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int)'
 );
 ^
In file included from c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:14,
                 from C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino:4:
c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:59:3: note: candidate: 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, int32_t, bool)'
   Arduino_ESP32RGBPanel(
   ^~~~~~~~~~~~~~~~~~~~~
c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:59:3: note:   candidate expects 31 arguments, 23 provided
c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:56:7: note: candidate: 'constexpr Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(const Arduino_ESP32RGBPanel&)'
 class Arduino_ESP32RGBPanel
       ^~~~~~~~~~~~~~~~~~~~~
c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:56:7: note:   candidate expects 1 argument, 23 provided
c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:56:7: note: candidate: 'constexpr Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(Arduino_ESP32RGBPanel&&)'
c:\Users\ftens\Sviluppo SW\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32RGBPanel.h:56:7: note:   candidate expects 1 argument, 23 provided
C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino:21:2: error: 'Arduino_RPi_DPI_RGBPanel' does not name a type; did you mean 'Arduino_ESP32RGBPanel'?
  Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  ^~~~~~~~~~~~~~~~~~~~~~~~
  Arduino_ESP32RGBPanel
C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino: In function 'void setup()':
C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino:58:3: error: 'gfx' was not declared in this scope
   gfx->begin();
   ^~~
C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino: In function 'void loop()':
C:\Users\ftens\Sviluppo SW\Arduino\_Arduino_GFX\MakerFabs_4_3\MakerFabs_4_3.ino:86:5: error: 'gfx' was not declared in this scope
     gfx->fillScreen(BLUE);
     ^~~
Più di una libreria trovata per "TAMC_GT911.h"
  Usata: C:\Users\ftens\Sviluppo SW\Arduino\libraries\TAMC_GT911
  Non usata: C:\Users\ftens\Sviluppo SW\Arduino\libraries\gt911-arduino-main
exit status 1

Compilation error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int)'

Can you help me ?

moononournation, 2023-02-23:
Sorry, I didn't have this board in hand. Please follow the new method signature to fix it.

SvenMb, 2023-03-01:
I use this, and it works with 1.3.2:
```
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
    0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel);
```

moononournation, 2023-03-02:
Tested this declaration ok:
```
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 40 /* hsync_front_porch */, 48 /* hsync_pulse_width */, 88 /* hsync_back_porch */,
    0 /* vsync_polarity */, 13 /* vsync_front_porch */, 3 /* vsync_pulse_width */, 32 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel);
```

copyCATandpasteDOG, 2023-03-03:
I tested it too and it works

```
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 40 /* hsync_front_porch */, 48 /* hsync_pulse_width */, 88 /* hsync_back_porch */,
    0 /* vsync_polarity */, 13 /* vsync_front_porch */, 3 /* vsync_pulse_width */, 32 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel);
```
But i have difficulties to build a touch calibrate sketch with the GT911 driver
does arduino_gfx support this?

SvenMb, 2023-03-03:
The gt911 doesn't really need calibration.

I changed the touch.h to the following: Also be aware that you have to initialize the touch **after** the gfx, since only then you have the correct size of the panel avaiable.

I also changed the behaviour of release, so long touches and moves work.

But be aware that the TAMC_gt911 driver seems to be buggy, since I sometimes need to restart the board to get touch working. If I find some time I will reimplement touch.h with the arduino-goodix driver, since that one seems to be stable in openHASP. 

```touch.h
/*******************************************************************************
 * Touch libraries:
 * GT911: https://github.com/TAMCTec/gt911-arduino.git
 ******************************************************************************/

#define TOUCH_GT911
#define TOUCH_GT911_SCL 20
#define TOUCH_GT911_SDA 19
#define TOUCH_GT911_INT -1
#define TOUCH_GT911_RST 38
#define TOUCH_GT911_ROTATION ROTATION_INVERTED
#define TOUCH_GT911_X 800
#define TOUCH_GT911_Y 480

int16_t touch_last_x = 0, touch_last_y = 0;

#include <TAMC_GT911.h>
TAMC_GT911 ts = TAMC_GT911(TOUCH_GT911_SDA, TOUCH_GT911_SCL, TOUCH_GT911_INT, TOUCH_GT911_RST, TOUCH_GT911_X, TOUCH_GT911_Y);

void touch_init(int max_x, int max_y)
{
  ts.begin();
  ts.setRotation(TOUCH_GT911_ROTATION);
  // only needed once, since it is stored in touch ctrl flash
  // ts.setResolution(TOUCH_GT911_Y, TOUCH_GT911_Y)
}

bool touch_has_signal()
{
  return true;
}

bool touched = false;
unsigned long lasttouch;

bool touch_touched()
{
  ts.read();
  if (ts.isTouched)
  {
    // Serial.printf(" X: %d\n",ts.points[0].x);
    // Serial.printf(" Y: %d\n",ts.points[0].y);
    touch_last_x = ts.points[0].x;
    touch_last_y = ts.points[0].y;
    touched=true;
    lasttouch = millis();
    return true;
  }
  return false;
}

bool touch_released()
{
  if (touched) {
    ts.read();
    if (ts.isTouched) {
      lasttouch = millis();
    } else if (lasttouch+80 < millis()) {
      // Serial.printf(" RX: %d\n",ts.points[0].x);
      // Serial.printf(" RY: %d\n",ts.points[0].y);

      touch_last_x = ts.points[0].x;
      touch_last_y = ts.points[0].y;
      touched=false;
      return true;
    }
  }
  return false;
}
```

moononournation, 2023-03-03:
The touch libraries in Arduino IDE is a little bit complicated, may be someone should unified most drivers just like Arduino_GFX unified display drivers.

copyCATandpasteDOG, 2023-03-03:
Thanks  very much SvenMb for help,but ooops,i didn't notice it was for a 4.3 inch tft,mine is 7inch 800x600,so i change the
resolution settings to 800x600,INT 25,RST 26 to adapt to the GT911 touchprint sketch that worked for me,but this one
gives me errors:

```
c:/users/utilisateur/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\UTILIS~1\AppData\Local\Temp\arduino_cache_929465\core\core_65426a1e881358a296d8f0f598f4c33e.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
c:/users/utilisateur/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\UTILIS~1\AppData\Local\Temp\arduino_cache_929465\core\core_65426a1e881358a296d8f0f598f4c33e.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
c:/users/utilisateur/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\UTILIS~1\AppData\Local\Temp\arduino_cache_929465\core\core_65426a1e881358a296d8f0f598f4c33e.a(main.cpp.o): in function `loopTask(void*)':
C:\Users\Utilisateur\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/main.cpp:42: undefined reference to `setup()'
c:/users/utilisateur/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\Utilisateur\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/main.cpp:48: undefined reference to `loop()'

```

My sketch prologue:

```
#define TOUCH_GT911
#define TOUCH_GT911_SCL 20
#define TOUCH_GT911_SDA 19
#define TOUCH_GT911_INT 25
#define TOUCH_GT911_RST 26
#define TOUCH_GT911_ROTATION ROTATION_INVERTED
#define TOUCH_GT911_X 800
#define TOUCH_GT911_Y 600

int16_t touch_last_x = 0, touch_last_y = 0;

#include <TAMC_GT911.h>
TAMC_GT911 ts = TAMC_GT911(TOUCH_GT911_SDA, TOUCH_GT911_SCL, TOUCH_GT911_INT, TOUCH_GT911_RST, TOUCH_GT911_X, TOUCH_GT911_Y);

```

SvenMb, 2023-03-03:
No I have the current 8048S070 mit capacitive touch. The pin settings work on this board.

SvenMb, 2023-03-03:
And what I provided is not a sketch it is a headerfile to substitute the touch.h which is used in the examples in this library.

copyCATandpasteDOG, 2023-03-03:
You're completly right,i blindly copy your snippet thinking it was a complete sketch :)
i works now.

closed 2023-03-04 by moononournation

6bigfire, 2023-04-09:
> Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
>     41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
>     14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */, [...]

it's wonderful!

copyCATandpasteDOG, 2023-04-09:
I forgot to mention that the touch works without the INT & RST settings:

```
 #define TOUCH_GT911
 #define TOUCH_GT911_SCL 20
 #define TOUCH_GT911_SDA 19
 #define TOUCH_GT911_INT -1
 #define TOUCH_GT911_RST -1
 #define TOUCH_GT911_ROTATION ROTATION_NORMAL
 #define TOUCH_MAP_X1 800
 #define TOUCH_MAP_X2 0
 #define TOUCH_MAP_Y1 480
 #define TOUCH_MAP_Y2 0

```