moononournation/Arduino_GFX #270: ESP32 S3 8MB PSRAM

aaron-neal, 2023-03-16:
I have a custom board with an RGB panel attached, in trying to bring up the board I cannot get this driver to run.
I get a crash after calling gfx->begin(), seems to be a watchdog maybe?--> rst:0x8 (TG1WDT_SYS_RST)

Do you have any debuggings suggestions that I could investigate?

Setup as below:

```Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    0 /* DE */, 3/* VSYNC */, 46 /* HSYNC */, 9 /* PCLK */,
    1 /* R0 */, 2 /* R1 */, 42 /* R2 */, 41 /* R3 */, 40 /* R4 */,
    39 /* G0 */, 38 /* G1 */, 45 /* G2 */, 48 /* G3 */, 47 /* G4 */, 21 /* G5 */,
    14 /* B0 */, 13 /* B1 */, 12 /* B2 */, 11 /* B3 */, 10 /* B4 */, 
    0 /* hsync_polarity */, 15 /* hsync_front_porch */, 4/* hsync_pulse_width */, 15 /* hsync_back_porch */,
    0 /* vsync_polarity */, 10 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 15 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 9000000 /* prefer_speed */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(800, 480, bus);```

Below is the debugging output at the moment

![image](https://user-images.githubusercontent.com/8297625/225643204-0b9df7cc-1fbd-4473-b0dd-37d7f8232430.png)

moononournation, 2023-03-16:
try switching PSRAM mode, e.g. QSPI/OSPI

aaron-neal, 2023-03-16:
Thanks for the suggestion, that appears to check out OK, tried all variations which did cause the PSRAM fail to init.

I have tracked it down to this line:

```
ESP_ERROR_CHECK(esp_lcd_new_rgb_panel(_panel_config, &_panel_handle));
```
even if I change to 

```
esp_err_t err = (esp_lcd_new_rgb_panel(_panel_config, &_panel_handle));
```

It never returns from this function and resets withough providing an error reason / message.

Still investigating.

aaron-neal, 2023-03-17:
A real head scratcher, I have now proven PSRAM is fine with a small PSRAM benchmarking routine.

moononournation, 2023-03-17:
wild guess are you used GPIO 35, 36, 37?

aaron-neal, 2023-03-17:
Not for the RGB panel, I have I2C on 35/36 and an input on 37

[image]

These function fine.

moononournation, 2023-03-17:
OPI PSRAM occupied these pins

aaron-neal, 2023-03-17:
Ahhh, I am sorry to have wasted your time. I will try and swap for a quad version.

closed 2023-03-17 by aaron-neal

moononournation, 2023-03-17:
Thats ok. I heard many people encountered this problem, so I can guess it.