moononournation/Arduino_GFX #273: ESP32-4827S043 flickering problem

Emmanuele75, 2023-03-17:
Hello,
for a few days now, I have been trying to get your Arduino_gfx library to work properly with the ESP32-4827S043 module (4.3 inch 480x272 display). I have configured the display parameters as you indicated, but I am experiencing image flickering (it is as if the images have varying brightness). With the same configuration, this does not happen with the LovyanGFX library. Additionally, I have tried on three other ESP32-4827S043 boards, and they all have the same issue. 
How can I solve this problem? 
Thanks)

moononournation, 2023-03-18:
Do you have the config of using this display using LovyanGFX?

Emmanuele75, 2023-03-18:
yes!
LovyanGFX:
`include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>
#include <lvgl.h>

class LGFX : public lgfx::LGFX_Device{
  lgfx::Bus_RGB     _bus_instance;
  lgfx::Panel_RGB   _panel_instance;
  lgfx::Light_PWM   _light_instance;
  lgfx::Touch_XPT2046 _touch_instance;

public:LGFX(void){
  auto cfg    = _bus_instance.config();
  cfg.panel   = &_panel_instance;
  cfg.pin_d0  = GPIO_NUM_8;  // B0
  cfg.pin_d1  = GPIO_NUM_3;  // B1
  cfg.pin_d2  = GPIO_NUM_46; // B2
  cfg.pin_d3  = GPIO_NUM_9;  // B3
  cfg.pin_d4  = GPIO_NUM_1;  // B4
  cfg.pin_d5  = GPIO_NUM_5;  // G0
  cfg.pin_d6  = GPIO_NUM_6;  // G1
  cfg.pin_d7  = GPIO_NUM_7;  // G2
  cfg.pin_d8  = GPIO_NUM_15; // G3
  cfg.pin_d9  = GPIO_NUM_16; // G4
  cfg.pin_d10 = GPIO_NUM_4;  // G5
  cfg.pin_d11 = GPIO_NUM_45; // R0
  cfg.pin_d12 = GPIO_NUM_48; // R1
  cfg.pin_d13 = GPIO_NUM_47; // R2
  cfg.pin_d14 = GPIO_NUM_21; // R3
  cfg.pin_d15 = GPIO_NUM_14; // R4
  cfg.pin_henable       = GPIO_NUM_40;
  cfg.pin_vsync         = GPIO_NUM_41;
  cfg.pin_hsync         = GPIO_NUM_39;
  cfg.pin_pclk          = GPIO_NUM_42;
  cfg.freq_write        = 12000000;   //16000000
  cfg.hsync_polarity    = 0; //0
  cfg.hsync_front_porch = 8; //8
  cfg.hsync_pulse_width = 4; //4
  cfg.hsync_back_porch  = 43; //43
  cfg.vsync_polarity    = 0;  //0
  cfg.vsync_front_porch = 8; //8
  cfg.vsync_pulse_width = 4; //4
  cfg.vsync_back_porch  = 12; //12
  cfg.pclk_idle_high    = 1; //1
  _bus_instance.config(cfg);
  _panel_instance.setBus(&_bus_instance);
  
  { auto cfg = _panel_instance.config();
  cfg.memory_width  = 480;
  cfg.memory_height = 272;
  cfg.panel_width   = 480;
  cfg.panel_height  = 272;
  cfg.offset_x      = 0;
  cfg.offset_y      = 0;
  _panel_instance.config(cfg);
  }
  
  { auto cfg = _panel_instance.config_detail();
  cfg.use_psram = 1;
  _panel_instance.config_detail(cfg);
  }

  { auto cfg = _light_instance.config();
  cfg.pin_bl = GPIO_NUM_2;
  _light_instance.config(cfg);
  }
  _panel_instance.light(&_light_instance);

{ 
      auto cfg = _touch_instance.config();
      cfg.x_min      = 100;    
      cfg.x_max      = 4000;   
      cfg.y_min      = 100;     
      cfg.y_max      = 4000;    
      cfg.pin_int    = -1;    
      cfg.bus_shared = true;  
      cfg.offset_rotation = 0;// 0-7
      cfg.spi_host = 1; // 
      cfg.freq = 1000000;    
      cfg.pin_sclk = 12;     
      cfg.pin_mosi = 11;     
      cfg.pin_miso = 13;     
      cfg.pin_cs   = 38;     
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  
    }
  setPanel(&_panel_instance); 
  }
};
LGFX tft; `

Arduino_GFX_Library:
`Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
   40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 8 /* 8 hsync_front_porch */, 4 /* 4 hsync_pulse_width */, 43 /* 43 hsync_back_porch */,
   0 /* vsync_polarity */, 8 /* 8 vsync_front_porch */, 4 /* 4 vsync_pulse_width */, 12 /* 12 vsync_back_porch */,
    1 /* pclk_active_neg */, 9000000 /* prefer_speed */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 272 /* height */, rgbpanel);`

Emmanuele75, 2023-03-18:
you can see that there is a difference in the prefer_speed (900000/12000000), changing this value from 8000000 to 12000000  don't feel any difference.

moononournation, 2023-03-18:
wild guess it is related to pclk_idle_high, could you please help to test again with latest code?

Emmanuele75, 2023-03-18:
Of course I'm willing to do all the tests you deem appropriate

Emmanuele75, 2023-03-21:
Any news?
Thanks!

moononournation, 2023-03-21:
just released v1.3.4, please try PDQgraphicstest

Emmanuele75, 2023-03-22:
I did as you said, but unfortunately it doesn't work.
Look at the pictures:
[image]

moononournation, 2023-03-22:
you may also try align the prefer_speed to 12000000

moononournation, 2023-03-22:
can you also test 0, 1 option?

Emmanuele75, 2023-03-22:
https://user-images.githubusercontent.com/128136690/226871742-c00831dc-38db-4015-aab4-83a52ccbfb20.mp4

https://user-images.githubusercontent.com/128136690/226871783-2b66ecab-00df-43e3-83ac-ea061333062d.mp4

https://user-images.githubusercontent.com/128136690/226871846-138ff90f-15fc-48b3-bb27-868150f6b7b3.mp4

moononournation, 2023-03-22:
any code?

Emmanuele75, 2023-03-22:
I do not understand

moononournation, 2023-03-22:
the prefer frequency

Emmanuele75, 2023-03-22:
```cpp
#pragma GCC optimize ("Ofast")

#define TFT_BL 2
#define CS_PIN  38
#define TIRQ_PIN -1// 18

#include <SPI.h>
#include <Arduino_GFX_Library.h>
#include <XPT2046_Touchscreen.h>
#include <lvgl.h>
#include "ui.h"

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 43 /* hsync_back_porch */,
    0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 12 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 9000000 /* prefer_speed */,
    0 /* de_idle_high */, 0 /* pclk_idle_high */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 272 /* height */, rgbpanel);

#include "touch.h"

/*Change to your screen resolution*/
static uint32_t screenWidth;
static uint32_t screenHeight;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t *disp_draw_buf;
static lv_disp_drv_t disp_drv;

void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p){
  uint16_t w = (area->x2 - area->x1 + 1);
  uint16_t h = (area->y2 - area->y1 + 1);
#if (LV_COLOR_16_SWAP != 0)
  gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#else
  gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#endif
  lv_disp_flush_ready(disp);
}

void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data){
  if (touch_has_signal()){
    if (touch_touched()){
      data->state = LV_INDEV_STATE_PR;
      data->point.x = touch_last_x;
      data->point.y = touch_last_y;
    }
    else if (touch_released()){
      data->state = LV_INDEV_STATE_REL;
    }
  }else{
    data->state = LV_INDEV_STATE_REL;
  }
}

void setup(){
  Serial.begin(115200); 
  ts.begin();
  ts.setRotation(0);
  gfx->begin();
  gfx->fillScreen(BLACK);
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);
  //analogWrite(TFT_BL, 100);
 
  lv_init();
  
  screenWidth = gfx->width();
  screenHeight = gfx->height();
  disp_draw_buf = (lv_color_t *)malloc(sizeof(lv_color_t) * screenWidth * 10);
  lv_disp_draw_buf_init(&draw_buf, disp_draw_buf, NULL, screenWidth * 10);
  

  /* Initialize the display */
  lv_disp_drv_init(&disp_drv);
  /* Change the following line to your display resolution */
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_drv_register(&disp_drv);

  /* Initialize the (dummy) input device driver */
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register(&indev_drv);

  ui_init();
}

void loop(){  
  lv_timer_handler(); 
  delay(5);
}
```

Emmanuele75, 2023-03-22:
I did the test, there is no difference with the previous tests :(

Emmanuele75, 2023-03-22:
with the LGFX library this doesn't happen, there will be a solution!

moononournation, 2023-03-22:
it should be

Emmanuele75, 2023-03-22:
anyway, when you want to do more tests I'm here :)
Thanks!

Frtrillo, 2023-04-17:
> I posted what works for me on these displays in lovyan gfx issues, arduino gfx actually works quite good. [lovyan03/LovyanGFX#374 (comment)](https://github.com/lovyan03/LovyanGFX/issues/374#issuecomment-1494590560)

It doesn't works properly for me. I get a lot of flickering on my UIs I ported from Squareline. On another display it works fine. Yes tho, the lvgl widgets demo works properly, its weird. It seems to work better with Lovyan but I still get some flickering on screen transitions. Its on the 5" display from sunton which uses gt911. I dont know if I should open a different issue

moononournation, 2023-04-17:
Please simply select the library you feel works better.

Zer0-bit, 2023-04-17:
> > I posted what works for me on these displays in lovyan gfx issues, arduino gfx actually works quite good. [lovyan03/LovyanGFX#374 (comment)](https://github.com/lovyan03/LovyanGFX/issues/374#issuecomment-1494590560)
> 
> It doesn't works properly for me. I get a lot of flickering on my UIs I ported from Squareline. On another display it works fine. Yes tho, the lvgl widgets demo works properly, its weird. It seems to work better with Lovyan example you posted but I still get some flickering on screen transitions. Its on the 5" display from sunton which uses gt911. I dont know if I should open a different issue

I'm using these same displays(though the 4.3" versions) with my own project and have no flickering issues, the flickering on your side is most likely you have some misconfiguration, different sizes displays will use different configuration parameters values, this isn't a library issue but  implementation issue on your side.

closed 2023-04-20 by moononournation