moononournation/Arduino_GFX #283: Arduino_NRFXSPI doesn't work with NRF52 boards

IgnatiousPei, 2023-04-14:
### Info:
- Arduino board: Adafruit Feather nRF52840 Sense
- Arduino IDE version: 2.0.4

### What I am trying to do:
I am trying to run the HelloWorld example using the NRFXSPI databus. 

### My code:
```cpp 
#include <Arduino_GFX_Library.h>

#define GFX_BL 0

Arduino_DataBus *bus = new Arduino_NRFXSPI(2 /* DC */, 7 /* CS */, 8 /* SCK */, 10 /* MOSI */, 9 /* MISO */);

Arduino_GFX *gfx = new Arduino_GC9A01(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);

void setup(void)
{
    gfx->begin();
    gfx->fillScreen(BLACK);

#ifdef GFX_BL
    pinMode(GFX_BL, OUTPUT);
    digitalWrite(GFX_BL, HIGH);
#endif

    gfx->setCursor(10, 10);
    gfx->setTextColor(RED);
    gfx->println("Hello World!");

    delay(5000); // 5 seconds
}

void loop()
{
    gfx->setCursor(random(gfx->width()), random(gfx->height()));
    gfx->setTextColor(random(0xffff), random(0xffff));
    gfx->setTextSize(random(6) /* x scale */, random(6) /* y scale */, random(2) /* pixel_margin */);
    gfx->println("Hello World!");

    delay(1000); // 1 second
}
```

###  My issue:
When I compile the code the following error is produced:
```sh
/home/ash/Arduino/sketches/test/test.ino:6:28: error: expected type-specifier before 'Arduino_NRFXSPI'
    6 | Arduino_DataBus *bus = new Arduino_NRFXSPI(2 /* DC */, 7 /* CS */, 8 /* SCK */, 10 /* MOSI */, 9 /* MISO */);
      |                            ^~~~~~~~~~~~~~~

exit status 1

Compilation error: expected type-specifier before 'Arduino_NRFXSPI'
```

### Attempted fixes on my end:

I tried to forcefully static define **ARDUINO_ARCH_NRF52840** in the file "_Arduino_GFX_Library.h_". This didn't work and gave a different error.  Maybe the error will be useful to you.

```sh

------------------------------------------
[INFO]  > Compiling sketch for  Seeeduino:nrf52:xiaonRF52840Sense .
In file included from /home/ash/devel/IggyWatch-standalone/xiao-arduino/.arduino/Arduino/libraries/GFX_Library_for_Arduino/src/Arduino_GFX_Library.h:24,
                 from /home/ash/devel/IggyWatch-standalone/xiao-arduino/xiao-arduino.ino:3:
/home/ash/devel/IggyWatch-standalone/xiao-arduino/.arduino/Arduino/libraries/GFX_Library_for_Arduino/src/databus/Arduino_mbedSPI.h:12:10: fatal error: drivers/SPIMaster.h: No such file or directory
   12 | #include "drivers/SPIMaster.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Next steps
@moononournation  I would like to clarify if I am doing anything wrong. Any assistance would be appreciated.
Additionally, I would like to know if this library supports the nRF52 boards fully or if it is a work in progress. If it is a work in progress I could try to create a PR with a fix, if i manage tot solve the issue.

moononournation, 2023-04-14:
sorry, I do not have nRF52 in hand so cannot support it.

IgnatiousPei, 2023-04-14:
No worries, friend. I shall look into this and if I find a fix, I will create a PR for this. Thank you.

closed 2023-06-10 by moononournation