moononournation/Arduino_GFX #284: Not displaying photo frame example using ESP32 and ST7735 display

laz048, 2023-04-15:
### What I have:

Board: [SparkFun Thing Plus - ESP32 WROOM (Micro-B)](https://www.sparkfun.com/products/15663)
Display: [Adafruit ST7735 160x128 Breakout](https://learn.adafruit.com/1-8-tft-display/pinouts)
Arduino IDE version: 2.0.4

### What I am trying to do:

Trying to replicate your photo frame build from this [Instructable](https://www.instructables.com/Face-Aware-OSD-Photo-Frame/). 

### What I have done:
1. Setup server using docker
2. Modified the WifiPhotoFrame.ino with my specific setup as follows:

**The wifi and server settings:**
```
const char *SSID_NAME = "MY_WIFI_SSID";
const char *SSID_PASSWORD = "MY_PASSWORD";

const char *HTTP_HOST = "192.168.1.160";                /* Your HTTP photo server host name */
const uint16_t HTTP_PORT = 8080;                        /* Your HTTP photo server port */
const char *HTTP_PATH_TEMPLATE = "/OSDPhoto?w=%d&h=%d"; /* Your HTTP photo server URL path template */
```

**ESP32 board data bus class:**

```
Arduino_DataBus *bus = new Arduino_ESP32SPI(27 /* DC */, 5 /* CS */, 18 /* SCK */, 23 /* MOSI */, GFX_NOT_DEFINED /* MISO */, VSPI /* spi_num */);

```
**ST7735 Display class:** (Swapped the width with height since my display is different, there is also a mention of "tab" colors, is this related to the display cover and tab color, because mine is green.)

```
Arduino_GFX *gfx = new Arduino_ST7735(
  bus, 7 /* RST */, 0 /* rotation */, false /* IPS */,
  160 /* width */, 128 /* height */,
  2 /* col offset 1 */, 1 /* row offset 1 */,
  2 /* col offset 2 */, 1 /* row offset 2 */);
```

### What I have verified:

1. Can connect to server from different clients (The same PC, and a smartphone)
2. STM32 board is connected to Wifi (checked the router connections)
3. Display and board both work with Adafruit example code from their library

### What is happening right now:

The screen is just blank, the backlight is on but no photos are actually shown on the display.
I'm not sure what the issue could be, does the HTTP_PATH_TEMPLATE need to be changed? Or am I using the wrong display and/or bus classes? Thank you for your help!

moononournation, 2023-04-16:
You should not swap the width and height value, use rotation instead.

laz048, 2023-04-16:
> You should not swap the width and height value, use rotation instead.

I tried all 4 rotation options [0-3], but still the same issue, the screen is just blank. Am I using the incorrect display class by any chance? Because you mention something about tab colors for the different variations of the ST7735. I am unsure as to which is mine.

moononournation, 2023-04-16:
you should try the PDQgraphicstest first