moononournation/Arduino_GFX #288: Esp32-S3  + ST7701 display hard faults after updating gfx version from 1.3.1 to latest

nrjn, 2023-04-24:
### Context

ST7701 display with ESP32-S3. 
- v1.3.1 - Works great

- v1.3.4 - Hard faults. Error message
`assert failed: lcd_ll_set_pixel_clock_prescale IDF/components/hal/esp32s3/include/hal/lcd_ll.h:109 (prescale <= LCD_LL_PCLK_DIV_MAX)
`

I tried with different configuration form `[PDQgraphicstest.ino](https://github.com/moononournation/Arduino_GFX/blob/master/examples/PDQgraphicstest/PDQgraphicstest.ino)` I couldn't fix this issue.

moononournation, 2023-04-25:
I am using latest Arduino_GFX version and ST7701 without problems, e.g.:
https://www.bilibili.com/video/BV17P411S7Vb

moononournation, 2023-04-29:
Please refer to latest PDQgraphicstest.ino for the code changed.

closed 2023-04-29 by moononournation

barnbrooksystems, 2023-07-29:
Using Makerfabs 7" display.  **GFX Library 1.3.1 works** with demo code.  GFX Library V1.3.7 does not it comes up with 'assert failed: lcd_ll_set_pixel_clock_prescale idf/components/hal/esp32s3/include/hal/lcd_ll.h:109 (prescale <= lcd_ll_pclk_div_max)' error and does not compile

moononournation, 2023-07-29:
Try latest PDQgraphicstest example