moononournation/Arduino_GFX #292: Arduino_AVRPAR16 doesn't display correctly on ILI9486 (MAR3513)

kennywclin, 2023-05-04:
Arduino_DataBus* bus = new Arduino_AVRPAR16(
  38 /* RS/DC */, 40 /* CS */, 39 /* WR */, GFX_NOT_DEFINED /* RD */, 3 /*portLow*/, 1 /*portHigh*/);
Arduino_GFX *gfx = new Arduino_ILI9486(bus, 41 /* RST */, 3);

I called these two functions trying to get MAR3513 working on an Arduino Mega2560 board (actually Mega 2560 + WiFi R3).
However, the display did not work correctly and showed nothing but a black block on a white background.
After series of tests, I found that elimination of the "writeC8D16D16Split" declaration and implementation will solve this problem.

void Arduino_AVRPAR16::writeC8D16D16Split(...)
{ .....
  _data16.value = d1;
  WRITE16(_data16.msb);
  WRITE16(_data16.lsb);
  .....
}

INLINE void Arduino_AVRPAR16::WRITE16(uint16_t d)
{ .....
  _data16.value = d;
  *_dataPortLow = _data16.lsb;
  *_dataPortHigh = _data16.msb;
  .....
}

In writeC8D16D16Split, the value of "_data16.value" will be changed after calling "WRITE16" since _data16 is a global variable.
The function of "writeC8D16D16Split" is also declared and implemented in the base class of "Arduino_AVRPAR16", "Arduino_DataBase", as follows.

void Arduino_DataBus::writeC8D16D16Split(uint8_t c, uint16_t d1, uint16_t d2)
{
  writeCommand(c);
  _data16.value = d1;
  write(_data16.msb);
  write(_data16.lsb);
  _data16.value = d2;
  write(_data16.msb);
  write(_data16.lsb);
}

The values of d1 and d2 kept unchanged after calling "write."  This test was done after removing "writeC8D16D16Split" in class Arduino_AVRPAR16.
It's queer!!! I cannot explain why!

Anyway, it goes fine without the declaration and implementation of ""writeC8D16D16Split" in class Arduino_AVRPAR16.

moononournation, 2023-05-04:
Thx for your findings. I committed a quick fix, please check is it can fix the problems.

kennywclin, 2023-05-05:
A new error issues.

e:\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_AVRPAR16.cpp: In member function 'virtual void Arduino_AVRPAR16::writeC8D16D16Split(uint8_t, uint16_t, uint16_t)':
e:\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_AVRPAR16.cpp:166:14: error: 'wrMaskBase' was not declared in this scope
   *_wrPort = wrMaskBase;
              ^~~~~~~~~~

exit status 1

Compilation error: exit status 1

kennywclin, 2023-05-05:
If deleting the declaration and implementation of "writeC8D16D16Split" in Arduino_AVRPAR16.h and Arduino_AVRPAR16.cpp is not the choice of correction, the following lines
      _data16.value = d1;
      WRITE16(_data16.msb);
      WRITE16(_data16.lsb);
    
      _data16.value = d2;
      WRITE16(_data16.msb);
      WRITE16(_data16.lsb);
may simpley be
      WRITE16(d1 >> 8);
      WRITE16(d1 & 0xff);
    
      WRITE16(d2 >> 8);
      WRITE16(d2 & 0xff);

It works well.

moononournation, 2023-05-05:
Yes, it can fix it. But some times later, if I see the code and forgotten this issue, I may change back and bug appear again.

kennywclin, 2023-05-06:
> Yes, it can fix it. But some times later, if I see the code and forgotten this issue, I may change back and bug appear again.

Don't worry, but I will keep mentioning it to you! Haaaa.....

moononournation, 2023-06-10:
assume you have tested ok and close this issue

closed 2023-06-10 by moononournation