moononournation/Arduino_GFX #294: Arduino_Canvas doesn't have a proper Destructor

h5n1xp, 2023-05-17:
While porting some software to run on top of Arduino_GFX, I noticed that deleting an Arduino_Canvas doesn't free the allocated framebuffer.

So repeatedly locating a canvas and then deleting them, causes the ps_ram to be used up very quickly.

moononournation, 2023-05-18:
Updated, please help to check.

h5n1xp, 2023-05-19:
Your class now works perfectly, but it seems the ESP32-S3's default memory allocator doesn't coalesce memory deallocations or something when free() is called... as on the Lillygo boards the freed ram isn't doesn't seem to be returned to the heap...

h5n1xp, 2023-05-23:
Accidentally found another issue, calling begin() twice will allocate another framebuffer, Perhaps there should be a guard against that.

moononournation, 2023-05-23:
ok, committed

closed 2023-05-24 by h5n1xp