moononournation/Arduino_GFX #295: Bug in Arduino_RGB_Display::draw16bitRGBBitmap

Lukaswnd, 2023-05-22:
Hi there, I just started using this library, so I'm not sure how it really works. But I think I found a bug, introduced in [this commit](https://github.com/moononournation/Arduino_GFX/commit/c37e7a825694e36c894934b84932cacf293dda2b). 

I am using a esp32s3 with 800*480 Panel with Parallel RGB-565 interface. Generally it works. I with this configuration.
But I'd like to use lvgl, so I just need one function to update the frame buffer

`Arduino_ESP32RGBPanel *panel = new Arduino_ESP32RGBPanel(
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
    0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 43 /* hsync_back_porch */,
    0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 12 /* vsync_back_porch */
);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, panel, 0, true);
  
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  int16_t w = (area->x2 - area->x1 + 1);
  int16_t h = (area->y2 - area->y1 + 1);

#if (LV_COLOR_16_SWAP != 0)
    gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#else
    gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#endif

  lv_disp_flush_ready(disp);
}`

In the void Arduino_RGB_Display::draw16bitRGBBitmap function, there is a Cache_WriteBack_Addr. I think the position could be calculated better and the size is wrong, but I'm really not sure, since I don't know how this lib works and I just changed some code there, because it looked wrong and my bug was fixed afterwards.

 I think you could decrease the cache_size by setting the cachePos, not only to the right line, but also to the right x coord. Next the size has to be calculated with the WIDTH not the HEIGHT, so WIDTH * h. I think the *2 is because there are uint16_t used, which are obviously 2 bytes wide. And I think you have to add the Width of the updated bitmap as well, because otherwise it's last row is missig. It might be possible, that this results in one row of the framebuffer to much. So it could also be cache_size = ((WIDTH-1) * h+ w) * 2;

`default: // case 0:
        cachePos = (uint32_t)(_framebuffer + (y * WIDTH) + x);
        cache_size = (WIDTH * h+ w) * 2;`

moononournation, 2023-05-22:
Thx reporting this first.
Want to clarify why `+ w` in `cache_size = (WIDTH * h+ w) * 2;`

moononournation, 2023-05-23:
ok, commited

moononournation, 2023-06-10:
assume you have tested ok and close this issue

closed 2023-06-10 by moononournation