moononournation/Arduino_GFX #296: Arduino_XL9535SWSPI

P3Fab, 2023-05-25:
Using Library with GC9A01.
Any version past 1.3.1 gives the following error when I try to compile, anything 1.3.1 and older will compile without issue. 

```
In file included from ~\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_XL9535SWSPI.h:4:0,
  from ~\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_XL9535SWSPI.cpp:1:
~\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.8\libraries\Wire\src/Wire.h: In member function 'uint8_t Arduino_XL9535SWSPI::readRegister(uint8_t, uint8_t*, uint8_t)':
~\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.8\libraries\Wire\src/Wire.h:63:12: note: candidate 1: size_t TwoWire::requestFrom(int, int)
     size_t requestFrom(int, int);
            ^~~~~~~~~~~
~\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.8\libraries\Wire\src/Wire.h:61:12: note: candidate 2: virtual size_t TwoWire::requestFrom(uint8_t, size_t)
     size_t requestFrom(uint8_t, size_t);

```

moononournation, 2023-05-26:
ok, committed a fix for megaavr. which megaavr board are you using?

P3Fab, 2023-05-26:
Arduino Nano Every

moononournation, 2023-06-10:
assume you have tested ok and close this issue

closed 2023-06-10 by moononournation

P3Fab, 2023-06-10:
Yes, after replacing the 2 Arduino_XL9535SWSP files it works.