moononournation/Arduino_GFX #320: Issue with GFX library on ESP32S3-N4R8 with arduino-esp32 >= 2.0.4

benlye, 2023-07-28:
I have recently bought one of [these devices](https://www.elecrow.com/esp32-display-7-inch-hmi-display-rgb-tft-lcd-touch-screen-support-lvgl.html). It has an ESP32S3-N4R8 and an RGB touch screen. 

It works well with arduino-esp32 v2.0.3 and Arduino_GFX 1.3.1, but as soon as I update to any newer version of arduino-esp32 the display behaves strangely - when initialized it fades in to a light grey and periodically showing something close to what should be on it.

I've been struggling with it for a few hours and need help to understand what has changed in the arduino-esp32 package so that I can work out how to fix it.  Do you have any suggestions?

This is my very minimal sketch I use just to test the screen.

```
#include <Arduino_GFX_Library.h>

#define TFT_BL 2

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
  9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);

Arduino_RPi_DPI_RGBPanel *lcd = new Arduino_RPi_DPI_RGBPanel(
  bus,
      800 /* width */, 0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
      480 /* height */, 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
      1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); // Init Display

  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);

  lcd->begin();
  lcd->fillScreen(BLACK);
  delay(2000);
  lcd->fillScreen(RED);
  delay(2000);
  lcd->fillScreen(GREEN);
  delay(2000);
  lcd->fillScreen(BLUE);
  delay(2000);
  lcd->fillScreen(WHITE);
  delay(2000);
  lcd->fillScreen(BLACK);
  delay(2000);
}

void loop() {
  // put your main code here, to run repeatedly:
  lcd->fillScreen(RED);
  delay(2000);
  lcd->fillScreen(GREEN);
  delay(2000);
  lcd->fillScreen(BLUE);
  delay(2000);
  lcd->fillScreen(WHITE);
  delay(2000);
  lcd->fillScreen(BLACK);
  delay(2000);
}
```

moononournation, 2023-07-28:
Please test the latest PDQgraphicstest first.

benlye, 2023-07-28:
Thanks, I've updated to Arduino GFX v1.3.7 and arduino-esp32 v2.0.11 and given that a go.  The test runs but the output on the screen looks more or less the same as my previous tests.  

This is what it looks like most of the time.  Once or twice I saw something like a shadow of the test output on the screen, but it's unpredictable.

[image]

This is my device declaration (added to `Arduino_GFX_dev_device.h`):
```
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
   9 /* G0 */, 46 /* G1 */,  3 /* G2 */,  8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */,  7 /* B1 */,  6 /* B2 */,  5 /* B3 */,  4 /* B4 */,
   0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
   0 /* vsync_polarity */,  22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
   1 /* pclk_active_neg */, 16000000 /* prefer_speed */
);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */
);
```

These are the benchmark numbers from the serial output:
```
Benchmark	micro-secs
Screen fill	147031
Text	31207
Pixels	772798
Lines	1072395
Horiz/Vert Lines	148597
Rectangles (filled)	516346
Rectangles (outline)	74086
Triangles (filled)	214236
Triangles (outline)	46009
Circles (filled)	87153
Circles (outline)	95309
Arcs (filled)	48846
Arcs (outline)	151876
Rounded rects (filled)	555484
Rounded rects (outline)	78499
Done!
```

I saw that you added support for the Wizee WZ8048C050 - my device is the 7" version of it, the WZ8048C070.  

I've noticed that on the manufacturer's documentation Wiki ([5 inch](https://www.elecrow.com/wiki/index.php?title=ESP32_Display_5.0%27%27_Intelligent_Touch_Screen_Wi-Fi%26BLE_800*480_HMI_Display) and [7 inch](https://www.elecrow.com/wiki/index.php?title=ESP32_Display_7.0%27%27_Intelligent_Touch_Screen_Wi-Fi%26BLE_800*480_HMI_Display)), both boards have a note saying to use arduino-esp32 v2.0.3. 

I wonder if you were able to successfully use/test the WZ8048C050 and what settings you used in the IDE?

benlye, 2023-07-28:
For what it's worth, I back-ported the latest PDQgraphicstest to work on my screen with Arduino_GFX v1.3.1 (which works with arduino-esp32 v2.0.3) and it seems to work ok, with the screen doing all the things that it seems like it should.

For the older library version, my devices are declared as:
```c
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
  9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
      800 /* width */, 0 /* hsync_polarity */, 210 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
      480 /* height */, 0 /* vsync_polarity */, 22 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */,
      1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);
```

moononournation, 2023-07-28:
Update both Arduino_GFX and arduino-esp32 to latest version should fix it.

moononournation, 2023-07-28:
I can run 5" board in latest code, just 2 pins different:
```
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    1 /* hsync_polarity */, 8 /* hsync_front_porch */, 1 /* hsync_pulse_width */, 32 /* hsync_back_porch */,
    1 /* vsync_polarity */, 8 /* vsync_front_porch */, 1 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 12000000 /* prefer_speed */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);
```

benlye, 2023-07-28:
Yeah, the schematics for both confirm that those pins are different.  I think I have the pins right because they work on the older core and library version.

I'm going to try to step through the changes between v2.0.3 and v2.0.4 of the core to see if I can pinpoint when it stops working.  I don't know what else to do.

moononournation, 2023-07-28:
I do not have this in hand so cannot check it. But it is ok keep using workable version.

benlye, 2023-07-28:
Hopefully you can get Elecrow to send you a sample ;-)

I'll fall back to the older version if I have to, but I'm working on a project where I want to support multiple displays, and the others work well with the latest versions.

By the way, this is one of the other displays I'm using, and it works very well:
https://www.elecrow.com/esp-terminal-with-esp32-3-5-inch-parallel-480x320-tft-capacitive-touch-display-rgb-by-chip-ili9488.html

I can give you the device declaration if you're interested in adding it.

moononournation, 2023-07-28:
Welcome to add new dev device support. Please provide the Youtube PDQgraphicstest video as well.

benlye, 2023-07-28:
I've found the commit where my 7" display breaks - https://github.com/espressif/arduino-esp32/commit/666c66d3d1c7437aaebaaadf2084927b7fc476c4

It's the change to EspressIf IDF v4.4, and there are a lot of changes :-(

benlye, 2023-07-28:
> I can run 5" board in latest code, just 2 pins different:
> 

Would you mind sharing your Arduino IDE settings for the 5" board?

benlye, 2023-07-28:
Ugh, I have my 7" screen working. It was the sync settings. The ones that Elecrow put in their sample code did not match up with the datasheet for the screen.  Switching to the datasheet values has got the screen working.

This is what I'm using now:
```c
Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
  9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
  bus,
      800 /* width */, 0 /* hsync_polarity */, 40 /* hsync_front_porch */, 48 /* hsync_pulse_width */, 40 /* hsync_back_porch */,
      480 /* height */, 0 /* vsync_polarity */, 1 /* vsync_front_porch */, 31 /* vsync_pulse_width */, 13 /* vsync_back_porch */,
      1 /* pclk_active_neg */, 12000000 /* prefer_speed */, true /* auto_flush */);
```

I'll submit a PR for this display once I'm confident that I have it dialed in correctly.

benlye, 2023-07-28:
Might have been a bit hasty - the screen "works" but the PDQ test does something strange... The screen is completely blank until the second pass through the test when it comes on and shows everything.  Have you seen something like this before? Do I need to add some extra init steps?

benlye, 2023-07-30:
I'm still stuck with this.  I can't figure out why the screen would work OK with Lovyan but not with Arduino_GFX - with Arduino_GFX it takes anywhere from 0-90sec to initialize and then works fine.

This sketch works perfectly in LovyanGFX:
```c
#include<Arduino.h>
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>

#define GFX_BL  2

class LGFX : public lgfx::LGFX_Device
{
public:

  lgfx::Bus_RGB     _bus_instance;
  lgfx::Panel_RGB   _panel_instance;

  LGFX(void)
  {
    {
      auto cfg = _panel_instance.config();
      cfg.memory_width  = 800;
      cfg.memory_height = 480;
      cfg.panel_width  = 800;
      cfg.panel_height = 480;
      cfg.offset_x = 0;
      cfg.offset_y = 0;
      _panel_instance.config(cfg);
    }

    {
      auto cfg = _bus_instance.config();
      cfg.panel = &_panel_instance;
      
      cfg.pin_d0  = GPIO_NUM_15; // B0
      cfg.pin_d1  = GPIO_NUM_7;  // B1
      cfg.pin_d2  = GPIO_NUM_6;  // B2
      cfg.pin_d3  = GPIO_NUM_5;  // B3
      cfg.pin_d4  = GPIO_NUM_4;  // B4
      
      cfg.pin_d5  = GPIO_NUM_9;  // G0
      cfg.pin_d6  = GPIO_NUM_46; // G1
      cfg.pin_d7  = GPIO_NUM_3;  // G2
      cfg.pin_d8  = GPIO_NUM_8;  // G3
      cfg.pin_d9  = GPIO_NUM_16; // G4
      cfg.pin_d10 = GPIO_NUM_1;  // G5
      
      cfg.pin_d11 = GPIO_NUM_14; // R0
      cfg.pin_d12 = GPIO_NUM_21; // R1
      cfg.pin_d13 = GPIO_NUM_47; // R2
      cfg.pin_d14 = GPIO_NUM_48; // R3
      cfg.pin_d15 = GPIO_NUM_45; // R4

      cfg.pin_henable = GPIO_NUM_41;
      cfg.pin_vsync   = GPIO_NUM_40;
      cfg.pin_hsync   = GPIO_NUM_39;
      cfg.pin_pclk    = GPIO_NUM_0;
      cfg.freq_write  = 16000000;

      cfg.hsync_polarity    = 0;
      cfg.hsync_front_porch = 40;
      cfg.hsync_pulse_width = 48;
      cfg.hsync_back_porch  = 40;
      
      cfg.vsync_polarity    = 0;
      cfg.vsync_front_porch = 1;
      cfg.vsync_pulse_width = 31;
      cfg.vsync_back_porch  = 13;

      cfg.pclk_active_neg   = 1;
      cfg.de_idle_high      = 0;
      cfg.pclk_idle_high    = 0;
      
      _bus_instance.config(cfg);
    }
    _panel_instance.setBus(&_bus_instance);
    setPanel(&_panel_instance);
  }
};

LGFX gfx;

void setup() {
  Serial.begin(115200); // Init Display
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
  gfx.begin();
}

void loop() {
  gfx.fillScreen(TFT_RED);
  delay(2000);
  gfx.fillScreen(TFT_GREEN);
  delay(2000);
  gfx.fillScreen(TFT_BLUE);
  delay(2000);
  gfx.fillScreen(TFT_WHITE);
  delay(2000);
  gfx.fillScreen(TFT_BLACK);
  delay(2000);
}
```

I think I have translated everything to Arduino_GFX correctly, but the display takes anywhere from 0-90sec to initialize (and then displays perfectly):

```c
#include <Arduino_GFX_Library.h>

#define GFX_BL 2

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
   9 /* G0 */, 46 /* G1 */,  3 /* G2 */,  8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */,  7 /* B1 */,  6 /* B2 */,  5 /* B3 */,  4 /* B4 */,
   0 /* hsync_polarity */, 40 /* hsync_front_porch */, 48 /* hsync_pulse_width */, 40 /* hsync_back_porch */,
   0 /* vsync_polarity */,  1 /* vsync_front_porch */, 31 /* vsync_pulse_width */, 13 /* vsync_back_porch */,
   1 /* pclk_active_neg */, 16000000 /* prefer_speed */,
   0 /* de_idle_high */, 0 /* pclk_idle_high */
);
   
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */
);
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); // Init Display

  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);

  gfx->begin();
}

void loop() {
  gfx->fillScreen(RED);
  delay(2000);
  gfx->fillScreen(GREEN);
  delay(2000);
  gfx->fillScreen(BLUE);
  delay(2000);
  gfx->fillScreen(WHITE);
  delay(2000);
  gfx->fillScreen(BLACK);
  delay(2000);
}
```

@moononournation do you have any ideas? I appreciate that you don't have the screen, but if you can point me in a direction I'm happy to keep trying/testing.

moononournation, 2023-07-30:
What Arduino IDE version and what arduino-esp32 are you using?

benlye, 2023-07-30:
> What Arduino IDE version and what arduino-esp32 are you using?

I've tried Arduino IDEs 1.8.19 and 2.1.1, I'm using arduino-esp32 2.0.11.

moononournation, 2023-07-30:
no idea, I do not have one in hand and 5" version works fine for me.

benlye, 2023-07-30:
> no idea, I do not have one in hand and 5" version works fine for me.

Can you imagine where you would start if you did have one in hand?

I've tried compiling in PlatformIO.  The display reliably starts up in ~0-20s.  The compiler throws these warnings, I wonder if the last one is significant?

```
In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:22,
                 from C:/Users/blye/OneDrive - MathWorks/Documents/PlatformIO/Projects/230727-164016-esp32-s3-devkitc-1/src/A-7.0-TOUCH.ino:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPI.h:17: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:16,
                 from C:/Users/blye/OneDrive - MathWorks/Documents/PlatformIO/Projects/230727-164016-esp32-s3-devkitc-1/src/A-7.0-TOUCH.ino:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:23,
                 from C:/Users/blye/OneDrive - MathWorks/Documents/PlatformIO/Projects/230727-164016-esp32-s3-devkitc-1/src/A-7.0-TOUCH.ino:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPIDMA.h:8: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 1024

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:22,
                 from C:/Users/blye/OneDrive - MathWorks/Documents/PlatformIO/Projects/230727-164016-esp32-s3-devkitc-1/src/A-7.0-TOUCH.ino:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPI.h:17: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 32

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:25,
                 from C:/Users/blye/OneDrive - MathWorks/Documents/PlatformIO/Projects/230727-164016-esp32-s3-devkitc-1/src/A-7.0-TOUCH.ino:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:12: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:23,
                 from C:/Users/blye/OneDrive - MathWorks/Documents/PlatformIO/Projects/230727-164016-esp32-s3-devkitc-1/src/A-7.0-TOUCH.ino:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPIDMA.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:22,
                 from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.cpp:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPI.h:17: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:16,
                 from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.cpp:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:23,
                 from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.cpp:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPIDMA.h:8: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 1024

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:22,
                 from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.cpp:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPI.h:17: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 32

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:25,
                 from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.cpp:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:12: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32

In file included from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.h:23,
                 from .pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/Arduino_GFX_Library.cpp:1:
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32SPIDMA.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024

.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
.pio/libdeps/elecrow-hmi-70/GFX Library for Arduino/src/databus/Arduino_ESP32QSPI.cpp:66:15: warning: narrowing conversion of '((Arduino_ESP32QSPI*)this)->Arduino_ESP32QSPI::<anonymous>.Arduino_DataBus::_dataMode' from 'int8_t' {aka 'signed char'} to 'uint8_t' {aka 'unsigned char'} inside { } [-Wnarrowing]
       .mode = _dataMode,
```

I've ordered the 5" so it will be interesting to compare.

moononournation, 2023-07-30:
Please don't use PlatformIO, use Arduino IDE 1.8.19

closed 2023-07-30 by moononournation

benlye, 2023-07-30:
OK, I'll stick with Arduino IDE, but why close the issue?  I'd really like to get this to work and am happy to follow any advice you can give.

moononournation, 2023-07-30:
try Arduino IDE first

benlye, 2023-07-30:
I have tried Arduino IDE - 1.8.19 and v2.1.1 - the issue is in both of those. 

I think maybe you only saw my very last comment, and not the ones before?  I was only trying PIO as an extra test.  I am more than happy to work on this issue only in the Arduino IDE.

superhac, 2023-08-06:
I had also been struggling getting the 7" Elecrow to work.  But my model number is WZ8048C050 (notice mine ends in "50" rather then "70").  I'm not sure of there is any significance in the two but I was able to get it working quite well with the following code:

```
 #include <Arduino_GFX_Library.h>
#define GFX_BL 2
#define GFX_BL 2

Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 0 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */,
    1 /* hsync_polarity */, 8 /* hsync_front_porch */, 1 /* hsync_pulse_width */, 32 /* hsync_back_porch */,
    1 /* vsync_polarity */, 8 /* vsync_front_porch */, 1 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 12000000 /* prefer_speed */);

Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); // Init Display

  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);

  gfx->begin();

}

void loop() {
  gfx->fillScreen(RED);
  delay(2000);
  gfx->fillScreen(GREEN);
  delay(2000);
  gfx->fillScreen(BLUE);
  delay(2000);
  gfx->fillScreen(WHITE);
  delay(2000);
  gfx->fillScreen(BLACK);
  delay(2000);
}
```  
It's a combination of @benlye program with @moononournation 5 inch display driver.   Once I did that it worked great.  So thanks to both of you! I'm running 1.37 of GFX_Library and 2.0.11 esp32.

benlye, 2023-08-06:
> But my model number is WZ8048C050 (notice mine ends in "50" rather then "70").

Are you 100% sure you have the 7" screen and not the 5"? Have you measured your screen? You have the part number and pins for the 5".

superhac, 2023-08-06:
@benlye You know what your right.  I went back and measured it and its 5".  Then and went back and checked what I ordered and it was 7" and they sent me a 5".  I didn't realize the difference in the end numbers till you said that.  Good catch.  Now I have to figure out how to return this one and get the 7".   I will change the above.

benlye, 2023-08-06:
Good luck! If you do get the 7" I'll be interested in how you get on getting it to work with Arduino_GFX. I gave up and am now using LovyanGFX with mine.

I posted my working Lovyan config in the Elecrow forums:
https://forum.elecrow.com/discussion/473/lovyangfx-settings-for-7-display

superhac, 2023-08-06:
Thanks.   Once I get this RMA'd and get the right one sent back I check it out.  I'm hoping the RMA won't be an issue.... (Fingers Crossed).