moononournation/Arduino_GFX #322: color accuracy issue with 16 bit (esp8266 , ILI9341)

FelixLamontagne2, 2023-07-29:
Hardware: 
- esp8266 nodemcu dev board
- ILI9341 display

Hi,

When I'm trying to render the color `rgb565(31, 62, 0)` I get a light shade of orange that kind of looks like `~rgb565(31, 45, 0)` (this is just an estimate, I don't know the actual color being rendered), but when I'm trying to render the color `rgb565(31, 63, 0)` now I get a bright yellow (which is the expected color). The color `rgb565(31, 62, 0)` should also be a bright yellow and not orange. 
I used the red channel to make it more obvious when testing, but using `rgb565(0, 62, 0)` and `rgb565(0, 63, 0)` shows the same issue with the first one being a dark green while the second one is a bright green even though they should both be a bright green.

The red and blue channel also have the same issue where there is a significant and unexpected difference between 31 and 30.

The [Adafruit ILI9341](https://github.com/adafruit/Adafruit_ILI9341) library does not have this issue which seems to indicate that my display/hardware setup is correct.

Strangely, when calling `Adafruit_ILI9341::begin()` after calling `Arduino_ILI9341::begin()` the issue is fixed so it seems like some init command is incorrect or missing? It really needs to be after, if `Adafruit_ILI9341::begin()` is called before `Arduino_ILI9341::begin()` then the fix doesn't work. I looked at the code of the begin functions to try to find the issue, but unfortunately I am too new with hardware to really understand what is going on.

Below is the minimum code to reproduce the issue:
```cpp
#include <Arduino_DataBus.h>
#include <Arduino_G.h>
#include <Arduino_GFX.h>
#include <Arduino_GFX_Library.h>
#include <Arduino_TFT.h>
#include <Arduino_TFT_18bit.h>
#include <gfxfont.h>

#include <Adafruit_ILI9341.h>

#define TFT_CS         D8
#define TFT_RST        D4
#define TFT_DC         D3

Arduino_ESP8266SPI bus = Arduino_ESP8266SPI(TFT_DC, TFT_CS);
Arduino_ILI9341 display1 = Arduino_ILI9341(&bus, TFT_RST, 0 /* rotation */, false /* IPS */);

Adafruit_ILI9341 display2 = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);

void setup() 
{
  display1.begin();
  // Uncomment to fix the issue
  //display2.begin();
}

void loop() 
{
  // Displays a shade of orange, but bright yellow is expected
  display1.fillScreen(65472); // rgb(31,62,0)
  delay(500);
  // Displays the expected bright yellow
  display1.fillScreen(65504); // rgb(31,63,0)
  delay(500);
}
```

The reason I'm trying to get this issue fixed instead of just using the Adafruit library is because yours is 3-4 times more performant which is important in my project (thermal camera).

If you want additional information or for me to test something feel free to ask.

moononournation, 2023-07-29:
May be related to the gamma setting in initial code, let me check it later.

moononournation, 2023-07-29:
synced with Adafruit_ILI9341 init code, please get update and check is it ok now.

FelixLamontagne2, 2023-07-29:
Issue is resolved, thanks!

moononournation, 2023-12-22:
reopen it due to #406

moononournation, 2023-12-22:
@FelixLamontagne2 can you recheck is your color accuracy fix still valid after fix the typo?

closed 2024-01-05 by moononournation