moononournation/Arduino_GFX #324: Trying to install from source results in errors

denifomenyu, 2023-08-01:
Default project Hello World.ino fails to compile!
```
Compiling libraries...
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:22,
                 from C:\Users\Home\AppData\Local\Temp\.arduinoIDE-unsaved202371-4252-i7hjjh.jq3ms\HelloWorld\HelloWorld.ino:21:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32SPI.h:17: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:16,
                 from C:\Users\Home\AppData\Local\Temp\.arduinoIDE-unsaved202371-4252-i7hjjh.jq3ms\HelloWorld\HelloWorld.ino:21:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32QSPI.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:23,
                 from C:\Users\Home\AppData\Local\Temp\.arduinoIDE-unsaved202371-4252-i7hjjh.jq3ms\HelloWorld\HelloWorld.ino:21:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32SPIDMA.h:8: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 1024
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:22,
                 from C:\Users\Home\AppData\Local\Temp\.arduinoIDE-unsaved202371-4252-i7hjjh.jq3ms\HelloWorld\HelloWorld.ino:21:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32SPI.h:17: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 32
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:25,
                 from C:\Users\Home\AppData\Local\Temp\.arduinoIDE-unsaved202371-4252-i7hjjh.jq3ms\HelloWorld\HelloWorld.ino:21:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_HWSPI.h:12: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/Arduino_GFX_Library.h:23,
                 from C:\Users\Home\AppData\Local\Temp\.arduinoIDE-unsaved202371-4252-i7hjjh.jq3ms\HelloWorld\HelloWorld.ino:21:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src/databus/Arduino_ESP32SPIDMA.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024

Compiling library "GFX Library for Arduino"
"C:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++

In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.h:22,
                 from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:1:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus/Arduino_ESP32SPI.h:17: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.h:16,
                 from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:1:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus/Arduino_ESP32QSPI.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.h:23,
                 from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:1:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus/Arduino_ESP32SPIDMA.h:8: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 1024
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.h:22,
                 from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:1:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus/Arduino_ESP32SPI.h:17: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 32
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.h:25,
                 from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:1:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus/Arduino_HWSPI.h:12: warning: "SPI_MAX_PIXELS_AT_ONCE" redefined
 #define SPI_MAX_PIXELS_AT_ONCE 32
 
In file included from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.h:23,
                 from c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\Arduino_GFX_Library.cpp:1:
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus/Arduino_ESP32SPIDMA.h:8: note: this is the location of the previous definition
 #define SPI_MAX_PIXELS_AT_ONCE 1024
```

moononournation, 2023-08-01:
Please get update and try again

denifomenyu, 2023-08-01:
There are fewer mistakes...
```
Compiling libraries...
Compiling library "GFX Library for Arduino"
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'virtual bool Arduino_ESP32QSPI::begin(int32_t, int8_t)':
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:55:3: warning: missing initializer for member 'spi_bus_config_t::data4_io_num' [-Wmissing-field-initializers]
   };
   ^
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:55:3: warning: missing initializer for member 'spi_bus_config_t::data5_io_num' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:55:3: warning: missing initializer for member 'spi_bus_config_t::data6_io_num' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:55:3: warning: missing initializer for member 'spi_bus_config_t::data7_io_num' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:55:3: warning: missing initializer for member 'spi_bus_config_t::intr_flags' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::dummy_bits' [-Wmissing-field-initializers]
   };
   ^
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:66:15: error: narrowing conversion of '((Arduino_ESP32QSPI*)this)->Arduino_ESP32QSPI::<anonymous>.Arduino_DataBus::_dataMode' from 'int8_t' {aka 'signed char'} to 'uint8_t' {aka 'unsigned char'} inside { } [-Werror=narrowing]
       .mode = _dataMode,
               ^~~~~~~~~
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::duty_cycle_pos' [-Wmissing-field-initializers]
   };
   ^
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::cs_ena_pretrans' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::cs_ena_posttrans' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::input_delay_ns' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::pre_cb' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:71:3: warning: missing initializer for member 'spi_device_interface_config_t::post_cb' [-Wmissing-field-initializers]
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'void Arduino_ESP32QSPI::POLL_START()':
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:521:13: error: unused variable 'ret' [-Werror=unused-variable]
   esp_err_t ret = spi_device_polling_start(_handle, _spi_tran, portMAX_DELAY);
             ^~~
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp: In member function 'void Arduino_ESP32QSPI::POLL_END()':
c:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino\src\databus\Arduino_ESP32QSPI.cpp:535:13: error: unused variable 'ret' [-Werror=unused-variable]
   esp_err_t ret = spi_device_polling_end(_handle, portMAX_DELAY);
             ^~~
cc1plus.exe: some warnings being treated as errors
"C:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.5\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/newlib/platform_include" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/include" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/port/xtensa/include" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/freertos/include/esp_additions" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/include" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.11/tools/sdk/esp32/include/esp_hw_support/include/soc" "-IC:\\Users\\Home\\App [... cut ...] 3A3891F6D547E4AD\\libraries\\GFX_Library_for_Arduino\\databus\\Arduino_ESP32S2PAR8.cpp.o"

Using library GFX Library for Arduino at version 1.3.7 in folder: C:\Users\Home\Documents\Arduino\libraries\GFX_Library_for_Arduino 
Using library SPI at version 2.0.0 in folder: C:\Users\Home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SPI 
Using library Wire at version 2.0.0 in folder: C:\Users\Home\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\Wire
```

closed 2023-08-02 by moononournation

denifomenyu, 2023-08-02:
Works great! Thank you)