moononournation/Arduino_GFX #328: Arduino_RGB_Display.draw16bitRGBBitmap missing?

holgerlembke, 2023-08-07:
I want to display png-graphics on one of those ESP32-S3 HMI 800x480 boards. I follow the ImgViewerPng-Example.

To show the problem I created the following condensed example. It will not compile. I do not understand why.

```
#include <Arduino_GFX_Library.h>
#include <PNGdec.h>

#define GFX_DEV_DEVICE ESP32_8048S070
#define GFX_BL 2
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
  14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
  9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
  15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */,
  0 /* hsync_polarity */, 180 /* hsync_front_porch */, 30 /* hsync_pulse_width */, 16 /* hsync_back_porch */,
  0 /* vsync_polarity */, 12 /* vsync_front_porch */, 13 /* vsync_pulse_width */, 10 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  800 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */);

int16_t w, h, xOffset, yOffset;
PNG png;

void PNGDraw(PNGDRAW *pDraw)
{
  uint16_t usPixels[320];
  uint8_t usMask[320];

  // Serial.printf("Draw pos = 0,%d. size = %d x 1\n", pDraw->y, pDraw->iWidth);
  png.getLineAsRGB565(pDraw, usPixels, PNG_RGB565_LITTLE_ENDIAN, 0x00000000);
  png.getAlphaMask(pDraw, usMask, 1);
  gfx->draw16bitRGBBitmap(xOffset, yOffset + pDraw->y, usPixels, usMask, pDraw->iWidth, 1);
}

void setup() {}

void loop() {}
```

holgerlembke, 2023-08-07:
Working solution is to unroll the code from another function into this. Like.

```
void DrawPng(PNGDRAW *pDraw) {
  uint16_t usPixels[320];
  uint8_t usMask[320];

  pngparam->png.getLineAsRGB565(pDraw, usPixels, PNG_RGB565_LITTLE_ENDIAN, 0x00000000);
  pngparam->png.getAlphaMask(pDraw, usMask, 1);

  int y = pDraw->y + yOffset;
  uint8_t byte = 0;
  gfx->startWrite();
  for (int16_t i = 0; i < pDraw->iWidth; i++) {
    if (i & 7) {
      byte <<= 1;
    } else {
      byte = usMask[i / 8];
    }
    if (byte & 0x80) {
      gfx->writePixel(xOffset + i, y, usPixels[i]);
    }
  }
  gfx->endWrite();
}
```

closed 2023-08-07 by holgerlembke

moononournation, 2023-08-14:
Please get update and try again.