moononournation/Arduino_GFX #338: mjpeg 'freezes' output but thinks its still playing?

ladyada, 2023-08-26:
hi there, thanks for writing this great library - its very fun to use!

i tried out the mjpeg decoder to 480x480 RGB TTL display, with SD set up as 1-bit MMC to get about 10 FPS - module is ESP32S3-16NR8 set up with OPI PSRAM

it works very well but then 'freezes' after 20 seconds, the decoder thinks its still playing frames but the display doesnt change. i looped the playback and it isnt that the display is frozen or crashed, after the file is done decoding, it starts over and the video plays correctly from the start 

does this sound familiar? i dont think im converting the mp4 incorrectly but perhaps there's something simple going on :)

thanks for any suggestions - i will also keep trying different settings and sizes

--------------

input MP4: https://www.pexels.com/video/a-butterfly-on-a-flower-2616637/

converted with: 
`ffmpeg -i "butterfly.mp4" -vf "reverse,setpts=0.5*PTS,fps=10,vflip,hflip,rotate=90,scale=-1:640:flags=lanczos,crop=480:480" -pix_fmt yuv420p -q:v 9 butterfly.mjpeg`

output file: (rename to mjpeg)
[image]

----------------

Output (can see freezing in last 2 seconds)

https://github.com/moononournation/Arduino_GFX/assets/1214161/6adfa062-23dd-49e4-b9b3-3bdeec121fe3

-----------------

Code running, its modified to loop the video

```
/*******************************************************************************
 * Motion JPEG Image Viewer
 * This is a simple Motion JPEG image viewer example
 * Image Source: https://www.pexels.com/video/earth-rotating-video-856356/
 * cropped: x: 598 y: 178 width: 720 height: 720 resized: 240x240
 * ffmpeg -i "Pexels Videos 3931.mp4" -ss 0 -t 20.4s -vf "reverse,setpts=0.5*PTS,fps=10,vflip,hflip,rotate=90,crop=720:720:178:598,scale=240:240:flags=lanczos" -pix_fmt yuv420p -q:v 9 earth.mjpeg
 *
 ******************************************************************************/
#define MJPEG_FILENAME "/earth.mjpeg"
#define MJPEG_BUFFER_SIZE (240 * 240 * 2 / 10) // memory for a single JPEG frame
// #define MJPEG_FILENAME "/earth128.mjpeg"
// #define MJPEG_BUFFER_SIZE (128 * 128 * 2 / 10) // memory for a single JPEG frame

#include <Arduino_GFX_Library.h>
#include "inits.h"

#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

Arduino_DataBus *bus = new Arduino_SWSPI(
    GFX_NOT_DEFINED /* DC */, 15 /* CS */,
    7 /* SCK */, 6 /* MOSI */, GFX_NOT_DEFINED /* MISO */);
    
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    2 /* DE */, 42 /* VSYNC */, 41 /* HSYNC */, 1 /* PCLK */,
    13 /* R0 */, 12 /* R1 */, 11 /* R2 */, 10 /* R3 */, 46 /* R4 */,
    5 /* G0 */, 45 /* G1 */, 48 /* G2 */, 47 /* G3 */, 21 /* G4 */, 14 /* G5 */,
    5 /* B0 */, 5 /* B1 */, 40 /* B2 */, 39 /* B3 */, 38 /* B4 */,
    1 /* hync_polarity */, 10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    1 /* vsync_polarity */, 10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, GFX_NOT_DEFINED /* RST */, TL021WVC02_init_operations, sizeof(TL021WVC02_init_operations));

//#include <SD.h>
#include <SD_MMC.h>

#include "MjpegClass.h"
static MjpegClass mjpeg;
File mjpegFile;
[... 166 lines cut ...]
01:10:51.473 -> .............................................................................................................................................................................................................................................................................................................MJPEG end
01:11:29.787 -> Total frames: 301
01:11:29.787 -> Time used: 38428 ms
01:11:29.787 -> Average FPS: 7.8
01:11:29.787 -> Read MJPEG: 3830 ms (10.0 %)
01:11:29.787 -> Decode video: 26082 ms (67.9 %)
01:11:29.787 -> Show video: 8515 ms (22.2 %)
01:11:29.787 -> MJPEG start
01:11:29.882 -> .............................................................................................................................................................................................................................................................................................................MJPEG end
01:12:08.198 -> Total frames: 301
01:12:08.198 -> Time used: 38423 ms
01:12:08.198 -> Average FPS: 7.8
01:12:08.198 -> Read MJPEG: 3823 ms (9.9 %)
01:12:08.198 -> Decode video: 26226 ms (68.3 %)
01:12:08.198 -> Show video: 8374 ms (21.8 %)
01:12:08.198 -> MJPEG start
01:12:08.340 -> .............................................................................................................................................................................................................................................................................................................MJPEG end
01:12:46.647 -> Total frames: 301
01:12:46.647 -> Time used: 38428 ms
01:12:46.647 -> Average FPS: 7.8
01:12:46.647 -> Read MJPEG: 3825 ms (10.0 %)
01:12:46.647 -> Decode video: 26099 ms (67.9 %)
01:12:46.647 -> Show video: 8503 ms (22.1 %)
```
(etc forever)

moononournation, 2023-08-26:
In MiniTV test, 320x240 can play 15-30 FPS. So 480x480 should able to play 5-10 FPS. You may try reduce FPS to 5 to check is it have any improvements.

ladyada, 2023-08-26:
aha, i tried various debugging and retries and figured out in the end that ffmpeg was not encoding long videos properly to mjpeg, the `ffmpeg setpts=0.5*PTS` command was causing it freeze the imagery halfway (i would think it would just end the mjpeg file). im not sure why it was in the original command, as i dont see any timestamps handling in the mjpeg decoder.

anyways here's a completed 480x480 video that can be used for testing
[image]

have you tried using double-buffering to improve tearing effect? 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html#double-frame-buffer-in-psram

closed 2023-08-26 by ladyada