moononournation/Arduino_GFX #340: About the RGB screen that drives the ST7701

Xk-w, 2023-08-29:
There was a problem when I used the Arduino_GFX-1.3.6 library for the RGB screen driver of the ST7701 chip, the screen backlight could light up normally but the screen did not use any other reaction, this was not the case when Arduino_GFX-1.3.1 version, but the color of the screen reaction in the Arduino_GFX-1.3.1 version was wrong

Here is my code:

[Arduino_GFX-1.3.1]:Version Arduino_GFX-1.3.1 runs successfully

[[[

#include <Arduino.h>
#include <Arduino_GFX_Library.h>
#include "pin_config.h"
#include "XL95x5_Driver.h"

extern XL95x5 Class_XL95x5;

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    14 /* CS */, 36 /* SCK */, 35 /* MOSI */,
    38 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 41 /* PCLK */,
    5 /* R0 */, 4 /* R1 */, 3 /* R2 */, 2 /* R3 */, 1 /* R4 */,
    11 /* G0 */, 10 /* G1 */, 9 /* G2 */, 8 /* G3 */, 7 /* G4 */, 6 /* G5 */,
    45 /* B0 */, 46 /* B1 */, 42 /* B2 */, 13 /* B3 */, 12 /* B4 */
);
Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
    bus, -1, 0,
    false, 480, 480,
    st7701_type2_init_operations, sizeof(st7701_type2_init_operations),
    false);

XL95x5 Class_XL95x5(XL95x5_IIC_ADDRESS, XL95x5_SDA, XL95x5_SCL);

void setup(void)
{
    Serial.begin(115200);
    Class_XL95x5.begin();
    Class_XL95x5.read_all_reg(); // Read all registers

    Class_XL95x5.portMode(XL95x5_PORT_0, OUTPUT); // Configure the XL95x5 full port mode
    Class_XL95x5.portMode(XL95x5_PORT_1, OUTPUT);

    Class_XL95x5.digitalWrite(XL95x5_GPIO_NUM_6, HIGH);
    Class_XL95x5.digitalWrite(XL95x5_GPIO_NUM_5, HIGH);

    gfx->begin();

    gfx->fillScreen(BLACK);
    Serial.println("6");
}

void loop()
{
    // for (int i = 0; i < 255; i++)
    // {
    //     ledcWrite(1, i);
    //     delay(50);
    // }
    // for (int i = 255; i > 0; i--)
    // {
    //     ledcWrite(1, i);
    //     delay(50);
    // }

    gfx->println("Ciallo");
    delay(2000);
}

]]]

[Arduino_GFX-1.3.6]:Version Arduino_GFX-1.3.6 failed to run

[[[

#include <Arduino.h>
#include <Arduino_GFX_Library.h>
#include "pin_config.h"
#include "XL95x5_Driver.h"

Arduino_DataBus *bus = new Arduino_ESP32SPI(
    -1 /* DC */, 14 /* CS */,
    36 /* SCK */, 35 /* MOSI */, -1 /* MISO */);
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
    38 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 41 /* PCLK */,
    5 /* R0 */, 4 /* R1 */, 3 /* R2 */, 2 /* R3 */, 1 /* R4 */,
    11 /* G0 */, 10 /* G1 */, 9 /* G2 */, 8 /* G3 */, 7 /* G4 */, 6 /* G5 */,
    45 /* B0 */, 46 /* B1 */, 42 /* B2 */, 13 /* B3 */, 12 /* B4 */,
    1 /* hsync_polarity */, 6 /* hsync_front_porch */, 18 /* hsync_pulse_width */, 24 /* hsync_back_porch */,
    1 /* vsync_polarity */, 4 /* vsync_front_porch */, 10 /* vsync_pulse_width */, 16 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
    480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
    bus, -1 /* RST */, st7701_type2_init_operations, sizeof(st7701_type2_init_operations));

XL95x5 Class_XL95x5(XL95x5_IIC_ADDRESS, XL95x5_SDA, XL95x5_SCL);

void setup(void)
{
    Serial.begin(115200);
    Class_XL95x5.begin();
    Class_XL95x5.read_all_reg(); // Read all registers

    Class_XL95x5.portMode(XL95x5_PORT_0, OUTPUT); // Configure the XL95x5 full port mode
    Class_XL95x5.portMode(XL95x5_PORT_1, OUTPUT);

    Class_XL95x5.digitalWrite(XL95x5_GPIO_NUM_6, HIGH);

    if (!gfx->begin())
    // if (!gfx->begin(80000000)) /* specify data bus speed */
    {
        Serial.println("gfx->begin() failed!");
    }
    gfx->fillScreen(GREENYELLOW);
    Serial.println("6");
}

void loop()
{
    // for (int i = 0; i < 255; i++)
    // {
    //     ledcWrite(1, i);
    //     delay(50);
    // }
    // for (int i = 255; i > 0; i--)
    // {
    //     ledcWrite(1, i);
    //     delay(50);
    // }

    gfx->println("Ciallo");
    delay(2000);
}

]]]

Did I get something wrong or did the Arduino_GFX-1.3.6 library no longer support driving the ST7701 chip?

moononournation, 2023-08-29:
Please see the latest version of PDQgraphicstest.ino first.

Xk-w, 2023-08-30:
> 

The rgb screen can be successfully displayed, but why is the image function "gfx ->draw24bitRGBBitmap (0, 0, gImage1, 480, 480);//BRG" displayed? There are several rows of pixels on the left that are on the right, and the pixels are slightly offset. Can these be fixed by modifying the parameters in "st7701 type1 init_operations"? How should I solve this problem

[image]

Xk-w, 2023-08-30:
> The rgb screen can be successfully displayed, but why is the image function "gfx ->draw24bitRGBBitmap (0, 0, gImage1, 480, 480);//BRG" displayed? There are several rows of pixels on the left that [...]

Just now, I found the cause of the problem and removed the option "including image header data" when I was using the mold software to extract the image, so it can be displayed normally

closed 2023-08-30 by Xk-w